happycube/ld-decode #434: .tbc source frame comparison tool

simoninns, 2020-01-25:
When preparing images for combination it can be difficult to know which disc sources come from the same master.  It's easy if you have the physical discs (and can check the mint-markings), but with .ldf files from different places things are more tricky.

It might be possible to have an openCV based tool that compares the similarity of frames between .tbc files to give an indication if combination is likely to succeed.

Gamnn, 2020-01-31:
Might be worth taking a peek at the code for the vapoursynth median plugin for an idea on how to do this. It has a frame compare function for matching up sources. With debug on, it displays how closely each additional source matches with the first.
[image]

https://github.com/dubhater/vapoursynth-median/blob/master/src/median.cpp

simoninns, 2020-09-16:
I don't see this as necessary any more; it's way too much work to replace what is after all a pretty simple manual check

closed 2020-09-16 by simoninns