happycube/ld-decode #435: .tbc stacking tool

simoninns, 2020-01-25:
There may be some advantages of having a specialised tool to 'stack' multiple .tbc images.  At the moment external tools can be used to provide median stacking; but they lack the intelligence possible from using the ld-decode-tools metadata information.  There is also the possibility that (with ld-diffdod and ld-dropout-correct processed images) other stacking methods (such as a pure average) may result in better results for some types of disc sets.

atsampson, 2020-01-26:
I've been thinking that ld-dropout-correct would be the appropriate place to stack multiple sources. The work you've done to support correction from other sources means it's already got all the information it needs to do stacking, and (oversimplifying!) it already handles the cases where 0 or 1 sources are available for an output sample, so it'd just need to be taught how to handle 2 or more.

I'm thinking something along the following lines:

- For each line in each input field...
- Gather the dropout start/end locations for this line from all the sources, and sort these by X position. (So you have a list of entries like "source 2 dropout starts at x=150".)
- Iterate through the sorted list of locations, keeping a count of how many sources are valid at each point.
- You've now got a range of output samples within the line to compute, and you know how many sources are available for it:
  - If 0, compute the samples using the best candidates from other lines as present.
  - If 1, copy the samples directly from that source as present. (So this stays nice and efficient for the common case of 1 not-very-dropout-y source.)
  - If 2 or more, compute the samples by stacking the sources -- e.g. computing the median.

We need to bear in mind that the video samples are gamma-corrected, so any operations upon them need to be gamma-aware - taking the median of an odd number of samples works fine (because it's just based on ordering), but if you're taking the mean or doing other kinds of interpolation then you need to convert to linear light first and back afterwards.

At present the code assumes that there's one copy of each field, and it can copy ranges between the two because it never copies from another dropout; it'd be easier to handle multiple sources if (immutable) input fields and output fields were separate. I'd be tempted to set up a SourceField struct with the metadata and data for each field so it's easy to pass the list of sources around.

(The other two things I've been wondering about re dropout correction are: soft transitions into/out of corrected sections, and interpolating corrections from multiple candidates rather than using a single candidate. Both of those would be possible with the above too, e.g. when you went from 3 to 2 sources available you could do a transition over a few pixels so you don't get an obvious hard edge.)

simoninns, 2020-01-26:
Personally I'd rather keep stacking as a separate tool simply because I'm a fan of having tools as simple and focused as possible.  However, I could be persuaded on the subject.

Your comment about gamma corrected has me curious though - in ld-diffdod I believe the comparison is linear - I did make a note earlier today that perhaps it should be log instead.  Can I get your opinion on the way it does the actual differential comparison, and also the luma clip detection?  The code should be a lot easier to read now as I've separated all of the steps into different procedures.

simoninns, 2020-08-30:
Tool added in latest commit.  Raise new issues if there are bugs.

closed 2020-08-30 by simoninns