happycube/ld-decode #442: Making ld-diffdod more effective

simoninns, 2020-01-25:
After running some tests against various discs using ld-diffdod it looks likely that there are multiple ways in which disc decode (from the perspective of how the image degrades) - as well as dropouts there seems to be a low-pass filter effect muting the chroma and introducing high-frequency chroma noise.

My theory is that it would be useful to have ld-diffdod perform differential comparison of images using the luma part of the signal only as it would improve the ability to combine discs that have degraded... the combined chroma/luma comparison only seems to work well for good discs with dropouts at the moment.

I could, of course, be completely wrong - but it's worth a stab to see what the possible result is.

simoninns, 2020-01-26:
Seems I am forgetting my own code here... it is already converting to luma only.   So a few issues; the luma conversion doesn't seem to be very good and the code structure is a total mess.

So the current plan is to fix the luma filter and restructure the core differential comparison code to make everything clearer...  then check if it is actually doing what it's supposed to do.

atsampson, 2020-01-26:
Has anyone ever tried to characterise the noise that gets introduced by disc rot? When doing MTF correction (and other kinds of EQ) it'd be nice to know how much noise is getting added in different frequency bands...

simoninns, 2020-01-26:
Might be a good idea to open an issue for just that @atsampson - I've just pushed a new version of ld-diffdod to the repo, so I'm closing this issue for now :)

closed 2020-01-26 by simoninns