happycube/ld-decode #452: Vroop.

Gamnn, 2020-02-04:
VROOOOP!

[image]

[offsetvroop.zip](https://github.com/happycube/ld-decode/files/4153417/offsetvroop.zip)

This vroop also causes a desync in ld-discmap, which I'll open a separate issue for.

Gamnn, 2020-02-04:
Slightly longer clip, and a clip from the same section from a capture that decodes normally.

[issue453-broke.zip](https://drive.google.com/open?id=1gRvWHyRD6X6q2Qp9ZrSP9U1rgXYkp044)
[issue453-notbroke.zip](https://drive.google.com/open?id=1-Gd5JZexgC-hYE8m3s5oncGwZp_QFUtR)

happycube, 2020-02-06:
This is caused by a player skip.  Looking at the demodulated raw data there actually *is* an early VSYNC.

https://drive.google.com/open?id=1mkqJQzFHZjis7GZDic94WGevtJF5_L38

Gamnn, 2020-02-10:
Not certain if ld-decode could handle player-induced problems any better, so closing for now.

closed 2020-02-10 by Gamnn