happycube/ld-decode #453: ld-discmap: Vrooped frame triggers false positive of CLV timecode offset frame.

Gamnn, 2020-02-04:
The vroop in #452 causes ld-discmap to mark and remove it as a "CLV timecode offset." This causes a desync with the other .tbcs.
Additionally, The debug line for the timecode offset routine is commented out, which made it a little difficult to track down where it occurred.

simoninns, 2020-02-04:
CLV timecode offsets are documented by IEC 60857-1986 amendment 2 section 10.1.10.

The specification states the formula for working out what the frame number is; Offset frame numbers are detected, but do not cause padding.  Take a look at the spec for details and re-report if (without the ld-decode corruption) the issue reoccurs.

closed 2020-02-04 by simoninns

Gamnn, 2020-02-04:
Hard to test, as ld-discmap seems to crash on short clips. Interestingly on the clip, it detects a gap whereas on the full one it just detects an offset.

```
Debug: DiscMap(Frames 28, disc type is CLV, video format is NTSC, detected 0 pulldown frames)
Info: Input TBC is CLV NTSC
Info: Checking for lead in and out frames...
Info: Removing 0 frames marked as lead in/out
Info: Correcting frame numbers using sequence analysis...
Info: Sequence analysis corrected 0 frame numbers
Info: Searching for duplicate frames
Debug: Building list of VBIs that have more than one entry in the discmap...
Debug: Sorting the duplicated frame list into numerical order...
Debug: Removing any repeated frame numbers from the duplicated frame list...
Debug: Found 0 VBI frame numbers with more than 1 entry in the discmap
Info: Removed 0 duplicate frames - disc map size now 28 frames
Info: Verifying frame numbers are present for all frames in the disc map...
Info: Verification successful
Info: Sorting the disc map into numerical frame order...
Info: Sorting complete
Info: Looking for sequence gaps to pad in the disc map...
Debug: Sequence break: Current VBI frame is 18962 next frame is 18964 gap of 1 frames
Info: Found 1 gaps representing 1 missing frames in the disc map
Info: Note: The disc map has been padded.  This means that there were missing frames
Info: that will be represented by black frames in the output video.
Info: After padding the disc map contains 29 frames
Info: Disc mapping process completed
Info: Writing output video and metadata information...
Debug: SourceVideo::open(): Called with field byte length = 478660
Debug: SourceVideo::open(): Successful - 57 fields available
Info: Saving target video frames...
Floating point exception (core dumped)
```
syslog: `kernel: [547476.899691] traps: ld-discmap[14855] trap divide error ip:558732a21cc2 sp:7ffc8f9448b0 error:0 in ld-discmap[5587329f7000+3a000]`

[issue453-broke.zip](https://drive.google.com/open?id=1gRvWHyRD6X6q2Qp9ZrSP9U1rgXYkp044)
[issue453-notbroke.zip](https://drive.google.com/open?id=1-Gd5JZexgC-hYE8m3s5oncGwZp_QFUtR)