happycube/ld-decode #456: ld-discmap treats player skips as CLV timecode offsets

Gamnn, 2020-02-06:
A follow-up on #453 

Two different copies (out of 5) of a CLV disc captured on two different players have triggered this issue, in different spots.

The disc should have 80700 frames, but when the player skips a frame it's not padded and comes out to 80699.

```
ld-discmap -m side4-eb.tbc
Info: LaserDisc mapping tool
Info: 
Info: Please note that disc mapping is not fool-proof - if you
Info: have a disc that does not map correctly run ld-discmap
Info: with the --debug option for more details about the process
Info: (and the additional information required for issue reporting
Info: to the ld-decode project).
Info: 
Info: Some early LaserDiscs do not provide frame numbering or
Info: time-code information and cannot be automatically mapped -
Info: if in doubt verify your source TBC file using the ld-analyse
Info: application.
Info: 
Info: Note that NTSC CAV pulldown support currently only handles
Info: discs that follow the standard 1-in-5 pulldown pattern.
Info: 
Info: Processing input metadata for side4-eb.tbc
Info: Input TBC is CLV NTSC
Info: Checking for lead in and out frames...
Info: Removing 1 frames marked as lead in/out
Info: Correcting frame numbers using sequence analysis...
Info: Sequence analysis corrected 5 frame numbers
Info: Searching for duplicate frames
Info: Removed 0 duplicate frames - disc map size now 80699 frames
Info: Verifying frame numbers are present for all frames in the disc map...
Info: Verification successful
Info: Sorting the disc map into numerical frame order...
Info: Sorting complete
Info: Looking for sequence gaps to pad in the disc map...
Info: No gaps found in the disc map
Info: There were 1 CLV timecode offsets in the disc map
Info: After padding the disc map contains 80699 frames
Info: Disc mapping process completed
Info: --maponly selected.  No output file will be written.
```

In this case, the player messed up at timecode 42:25.13, and skips to 42.25.15, but ld-discmap does not pad the missing .14.

in ld-decode:
```
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
ERROR:root:Skipped field
file frame 5 CLV timecode 42:25.13 frame 76363
file frame 6 CLV timecode 42:25.16 frame 76366
```

[image]

For some reason, it will not reproduce from short clips. It properly detects a gap and pads them.
If the full .TBCs or .LDFs are needed, contact me on IRC.

[ebskip.ldf.zip](https://drive.google.com/open?id=1TkUZQDm0m52n86F7XTW_8t9mTc_XzHum)

simoninns, 2020-02-07:
The linked ldf does not trigger the error.  You are welcome on the project discord server to provide a link to a file that does (or link it here), then I can look into it.

Gamnn, 2020-02-07:
I can email a link. Would you prefer tbc or ldf?

simoninns, 2020-09-08:
Please retest and, if it's still failing, open a new issue and attach sample from recent ld-decode version for test/analysis; thanks.

closed 2020-09-08 by simoninns