happycube/ld-decode #459: Include PAL phase metadata in ld-decode

simoninns, 2020-02-07:
If there was phaseID metadata for PAL (as there is currently for NTSC) this would greatly assist ld-discmap since it would be possible to implement a common check to spot if skips are skips or simply corrupt VBI.  This would help with issues such as #456 and #451 as it would be worth while implementing if it could be applied to PAL and NTSC discs rather than just NTSC.

simoninns, 2020-07-30:
@happycube - any chance you could take a look at this?  It would help with some of the other issues and simplify the mapping code.

oyvindln, 2020-07-30:
I need to do burst phase detection for determining video track automatically for tapes, so planning to implement it in the python front-end. Will probably port from the C++ chroma decoder. If it's not already implemented by then I can submit it for inclusion for use for this as well.

atsampson, 2020-07-30:
Another use for this would be so transform3d knows if the phase sequence is wrong - at present it'll produce the wrong result, but it could limit the size of the FFT in those cases (or possibly fall back to 2D).

happycube, 2020-09-06:
Done.  (but needs more testing to make sure it gets everything right :) )

happycube, 2020-09-27:
Done, at least aside from rot side effects.

closed 2020-09-27 by happycube