happycube/ld-decode #460: ld-discmap fails to compile on mac/clang

Gamnn, 2020-02-11:
Getting numerous errors when I try to build ld-discmap on Mac.

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -O2 -std=gnu++11  -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.12 -Wall -W -fPIC -DQT_DEPRECATED_WARNINGS -DAPP_BRANCH="\"master\"" -DAPP_COMMIT="\"e14594d\"" -DQT_NO_DEBUG -DQT_CORE_LIB -I. -I../library/tbc -I/usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers -I. -I/usr/local/Cellar/qt/5.13.2/mkspecs/macx-clang -F/usr/local/Cellar/qt/5.13.2/lib -o discmap.o discmap.cpp
In file included from discmap.cpp:25:
In file included from ./discmap.h:28:
In file included from /usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers/QCoreApplication:1:
In file included from /usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers/qcoreapplication.h:43:
In file included from /usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers/qglobal.h:142:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:714:71: error: 
      invalid operands to binary expression ('const Frame' and 'const Frame')
    bool operator()(const _T1& __x, const _T1& __y) const {return __x < __y;}
                                                                  ~~~ ^ ~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3905:17: note: 
      in instantiation of member function 'std::__1::__less<Frame,
      Frame>::operator()' requested here
            if (__comp(*--__last, *__first))
                ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:4091:5: note: 
      in instantiation of function template specialization
      'std::__1::__sort<std::__1::__less<Frame, Frame> &, Frame *>' requested
      here
    __sort<_Comp_ref>(__first, __last, __comp);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:4100:12: note: 
      in instantiation of function template specialization 'std::__1::sort<Frame
      *, std::__1::__less<Frame, Frame> >' requested here
    _VSTD::sort(__first, __last, __less<typename iterator_traits<_Random...
           ^
discmap.cpp:512:10: note: in instantiation of function template specialization
      'std::__1::sort<Frame *>' requested here
    std::sort(m_frames.begin(), m_frames.end());
         ^
In file included from discmap.cpp:25:
In file included from ./discmap.h:28:
In file included from /usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers/QCoreApplication:1:
In file included from /usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers/qcoreapplication.h:43:
In file included from /usr/local/Cellar/qt/5.13.2/lib/QtCore.framework/Headers/qglobal.h:142:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3692:20: error: 
      no matching function for call to '__sort3'
    unsigned __r = __sort3<_Compare>(__x1, __x2, __x3, __c);
                   ^~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3912:20: note: 
      in instantiation of function template specialization
      'std::__1::__sort4<std::__1::__less<Frame, Frame> &, Frame *>' requested
      here
            _VSTD::__sort4<_Compare>(__first, __first+1, __first+2, --__...
                   ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:4091:5: note: 
      in instantiation of function template specialization
      'std::__1::__sort<std::__1::__less<Frame, Frame> &, Frame *>' requested
      here
    __sort<_Comp_ref>(__first, __last, __comp);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:4100:12: note: 
      in instantiation of function template specialization 'std::__1::sort<Frame
      *, std::__1::__less<Frame, Frame> >' requested here
    _VSTD::sort(__first, __last, __less<typename iterator_traits<_Random...
           ^
discmap.cpp:512:10: note: in instantiation of function template specialization
      'std::__1::sort<Frame *>' requested here
    std::sort(m_frames.begin(), m_frames.end());
         ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3652:1: note: 
      candidate template ignored: substitution failure [with _Compare =
      std::__1::__less<Frame, Frame> &, _ForwardIterator = Frame *]
__sort3(_ForwardIterator __x, _ForwardIterator __y, _ForwardIterator __z...
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3718:20: error: 
      no matching function for call to '__sort4'
[... 107 lines cut ...]
        _VSTD::__sort5<_Compare>(__first, __first+1, __first+2, __first+...
        ^~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__config:392:15: note: 
      expanded from macro '_VSTD'
#define _VSTD std::_LIBCPP_NAMESPACE
              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3715:1: note: 
      candidate template ignored: substitution failure [with _Compare =
      std::__1::__less<Frame, Frame> &, _ForwardIterator = Frame *]
__sort5(_ForwardIterator __x1, _ForwardIterator __x2, _ForwardIterator __x3,
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3827:5: error: 
      no matching function for call to '__sort3'
    __sort3<_Compare>(__first, __first+1, __j, __comp);
    ^~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:3652:1: note: 
      candidate template ignored: substitution failure [with _Compare =
      std::__1::__less<Frame, Frame> &, _ForwardIterator = Frame *]
__sort3(_ForwardIterator __x, _ForwardIterator __y, _ForwardIterator __z...
^
8 errors generated.
make[2]: *** [discmap.o] Error 1
make[1]: *** [sub-ld-discmap-make_first] Error 2
make: *** [build-tools] Error 2
```

atsampson, 2020-02-11:
Does making Frame::operator< a const member function help? i.e.

```
diff --git a/tools/ld-discmap/frame.cpp b/tools/ld-discmap/frame.cpp
index 9598562..2ee506f 100644
--- a/tools/ld-discmap/frame.cpp
+++ b/tools/ld-discmap/frame.cpp
@@ -168,7 +168,7 @@ void Frame::secondField(qint32 value)
 }
 
 // Overide less than operator for sorting
-bool Frame::operator<(const Frame& other)
+bool Frame::operator<(const Frame& other) const
 {
     return (m_vbiFrameNumber < other.m_vbiFrameNumber) ||
             ((m_vbiFrameNumber == other.m_vbiFrameNumber) && (other.m_isPullDown));
diff --git a/tools/ld-discmap/frame.h b/tools/ld-discmap/frame.h
index 9356fce..70f3c7c 100644
--- a/tools/ld-discmap/frame.h
+++ b/tools/ld-discmap/frame.h
@@ -66,7 +66,7 @@ public:
     void secondField(qint32 value);
 
     // Operators
-    bool operator <(const Frame &);
+    bool operator <(const Frame &) const;
 
 private:
     qint32 m_seqFrameNumber;
```

Gamnn, 2020-02-11:
Yup, that solves it.

PR #461 (2020-02-11): Make all operator member functions const
LLVM libcxx's std::sort requires operator< to be const, but it's good practice to do this for all of them.

Fixes #460.
closed 2020-02-11 by simoninns