happycube/ld-decode #466: ld-discmap - problem with dropout in VBI on CAV pulldown disc

Gamnn, 2020-02-17:
Some dropouts on this disc cause the VBI to decode as CLV, and seems to break ld-discmap. Running it with strict checking gives unmappable frames, running it without causes unnecessary padding.

[pulldownbreak.ldf](https://drive.google.com/open?id=1aOH9cp2z-9FOPC7qbYjSUvrdGzHrGLnI)

Strict:
```
Info: Processing input metadata for test.tbc
Debug: LdDecodeMetaData::read(): Loading JSON file "test.tbc.json"
Debug: Checking first 49 sequential frames for disc CAV/CLV type determination
Debug: Got 38 valid CAV picture numbers from 49 frames - source disc type is CAV
Debug: Storing VBI CAV picture numbers as frame numbers
Debug: Disc type is NTSC CAV - checking for pull-down frames
Debug: Seq. frame 16 looks like a pull-down, but there is no pull-down sequence in the surrounding frames - marking as false-positive
Debug: Seq. frame 32 looks like a pull-down, but there is no pull-down sequence in the surrounding frames - marking as false-positive
Debug: Performing a frame quality analysis for each frame
Debug: Frame: 0 60.6106 100 100 100 quality = 99.8612
Debug: Frame: 1 58.424 100 100 100 quality = 99.859
Debug: Frame: 2 61.3556 100 100 100 quality = 99.8619
Debug: Frame: 3 63.6405 80 100 100 quality = 99.8443
Debug: Frame: 4 59.8737 100 100 100 quality = 99.8604
Debug: Frame: 5 58.424 100 100 100 quality = 99.859
Debug: Frame: 6 58.424 100 100 100 quality = 99.859
Debug: Frame: 7 59.1449 100 100 100 quality = 99.8597
Debug: Frame: 8 60.6106 80 100 100 quality = 99.8412
Debug: Frame: 9 62.1089 100 100 100 quality = 99.8627
Debug: Frame: 10 54.9371 100 100 100 quality = 99.8555
Debug: Frame: 11 62.1089 100 100 100 quality = 99.8627
Debug: Frame: 12 62.8705 100 100 100 quality = 99.8634
Debug: Frame: 13 58.424 80 100 99.9993 quality = 99.8384
Debug: Frame: 14 60.6106 100 100 100 quality = 99.8612
Debug: Frame: 15 59.8737 100 100 99.9805 quality = 99.841
Debug: Frame: 16 62.1089 100 100 99.9687 quality = 99.8314
Debug: Frame: 17 62.1089 100 100 99.9756 quality = 99.8384
Debug: Frame: 18 62.1089 80 100 99.9763 quality = 99.8192
Debug: Frame: 19 59.1449 100 100 99.9721 quality = 99.832
Debug: Frame: 20 59.1449 100 100 99.9617 quality = 99.8216
Debug: Frame: 21 57.006 100 100 99.9714 quality = 99.8291
Debug: Frame: 22 69.2725 100 100 99.961 quality = 99.831
Debug: Frame: 23 59.1449 80 100 99.9589 quality = 99.7989
Debug: Frame: 24 63.6405 100 100 99.9568 quality = 99.8212
Debug: Frame: 25 64.419 100 100 99.9575 quality = 99.8226
Debug: Frame: 26 60.6106 100 100 99.9568 quality = 99.8182
Debug: Frame: 27 61.3556 100 100 99.9582 quality = 99.8203
Debug: Frame: 28 64.419 80 100 99.961 quality = 99.8062
Debug: Frame: 29 54.9371 100 100 99.9777 quality = 99.8333
Debug: Frame: 30 60.6106 80 100 99.9749 quality = 99.8163
Debug: Frame: 31 54.2626 100 100 99.9833 quality = 99.8382
Debug: Frame: 32 60.6106 100 100 99.9659 quality = 99.8272
Debug: Frame: 33 62.1089 80 100 99.9728 quality = 99.8157
Debug: Frame: 34 62.1089 100 100 99.9666 quality = 99.8294
Debug: Frame: 35 66.0017 100 100 99.9714 quality = 99.8381
Debug: Frame: 36 64.419 100 100 99.9819 quality = 99.8469
Debug: Frame: 37 59.1449 100 100 99.9819 quality = 99.8417
Debug: Frame: 38 57.006 80 100 100 quality = 99.8377
Debug: Frame: 39 59.1449 100 100 100 quality = 99.8597
Debug: Frame: 40 60.6106 100 100 100 quality = 99.8612
Debug: Frame: 41 61.3556 100 100 100 quality = 99.8619
Debug: Frame: 42 63.6405 100 100 100 quality = 99.8642
Debug: Frame: 43 61.3556 80 100 100 quality = 99.842
Debug: Frame: 44 67.6193 100 100 100 quality = 99.8681
Debug: Frame: 45 62.1089 100 100 100 quality = 99.8627
Debug: Frame: 46 61.3556 100 100 100 quality = 99.8619
Debug: Frame: 47 64.419 100 100 100 quality = 99.865
Debug: Frame: 48 54.9371 0 100 100 quality = 99.7559
Debug: DiscMap(Frames 49, disc type is CAV, video format is NTSC, detected 9 pulldown frames)
Info: Input TBC is CAV NTSC
Info: Checking for lead in and out frames...
Info: Removing 0 frames marked as lead in/out
Info: Correcting frame numbers using sequence analysis...
Debug: Broken VBI frame number sequence detected:
[... 126 lines cut ...]
Info: Searching for duplicate frames
Debug: Building list of VBIs that have more than one entry in the discmap...
Debug: Sorting the duplicated frame list into numerical order...
Debug: Removing any repeated frame numbers from the duplicated frame list...
Debug: Found 0 VBI frame numbers with more than 1 entry in the discmap
Info: Removed 0 duplicate frames - disc map size now 49 frames
Info: Numbering pulldown frames in the disc map...
Info: Numbering complete
Info: Verifying frame numbers are present for all frames in the disc map...
Info: Verification successful
Info: Sorting the disc map into numerical frame order...
Info: Sorting complete
Info: Looking for sequence gaps to pad in the disc map...
Debug: Sequence break over pulldown: Current VBI frame is 21201 next frame (+1) is 21201 gap of 0 frames
Debug: Sequence break over pulldown: Current VBI frame is 21214 next frame (+1) is 21216 gap of 2 frames
Info: Found 2 gaps representing 2 missing frames in the disc map
Info: Note: The disc map has been padded.  This means that there were missing frames
Info: that will be represented by black frames in the output video.
Info: After padding the disc map contains 51 frames
Info: Searching disc map for pulldown frames...
Info: Search complete; pulldown frames are present.  Renumbering disc map...
Info: Renumbering complete
Info: Disc mapping process completed
Info: --maponly selected.  No output file will be written.
```

Gamnn, 2020-02-17:
By manually correcting the vbiData on frame 16 from: ` [9297920,16388610,6388610]` to ` [9289728,16388610,6388610]`, it was able to map properly, even without correcting the vbiData for frame 32.

simoninns, 2020-07-30:
ld-discmap has to rely on VBI information to work... if the VBI is wrong, it might fail.  I'd say this was an upstream issue... perhaps ld-process-vbi would have corrected VBI which ld-decode failed to get correctly (you don't state if that had been run).  Either way, I don't think this is an issue with ld-discmap - and there is already another issue to do with editing VBIs.

closed 2020-07-30 by simoninns