happycube/ld-decode #468: CX tuning

Gamnn, 2020-02-19:
CX-encoded analog audio still sounds much worse from ld-decode than it does from a player.

Work was started with #138, but it still needs some tuning. Let me know if there are any kinds of samples that would be useful for such.

happycube, 2022-01-04:
I fixed the worst of it, but I don't have a working 'fast' filter so it's not a very good expander yet.

happycube, 2022-01-17:
Basically to do this right it needs to be made equivalent to the actual circuit.  If someone can help out here I'd appreciate it.

(I think there was some gnuradio work done - also if there's a SPICE simulation and a way to compile it into Python or C...)

simoninns, 2025-12-10:
You are welcome to open a new issue - I'm trying to clean up things as it's a bit difficult to handle 70 issues ranging back years.  A lot are just issues just hanging.  I would split this is two; new one issue to move whatever the broken tool is to the new home for them "prototypes/" and another feature request for what's actually needed. (to keep things actionable)