happycube/ld-decode #471: Dropout Blues - False positive on the dropout detection.

IanSmallshire, 2020-03-03:
[image]

[image]

[image]

[image]

Four samples supplied here:
https://drive.google.com/open?id=1MQ5BFlAmXey_C2TYhc4qnm8l7phxvzPA

happycube, 2020-03-08:
Found the check that caused this and disabled it... should be good now?

IanSmallshire, 2020-03-08:
Will kick off decode of the full disc again and update

IanSmallshire, 2020-04-15:
Much better than it was with almost perfect decode.

closed 2020-04-17 by happycube