happycube/ld-decode #474: Add VITC support to ld-process-vbi

atsampson, 2020-03-09:
Some PAL discs include [vertical interval timecode](https://tech.ebu.ch/publications/tech3097), presumably as a by-product of the mastering process. 

[image]

ld-process-vbi could decode this and store it in the JSON.

simoninns, 2021-12-13:
are there any disc samples available for implementing this?  (or a list of discs known to contain it).

Even better; some .lds files on the project SFTP.

Otherwise this one's for closing since there doesn't seem to be any viable implementation.

atsampson, 2021-12-13:
The PAL discs I've noticed it on, all on line 21 (41/42 in ld-analyse's numbering), are:

- EE Fawlty Towers
- Louvre 1 (haven't checked the other discs)
- EE Jason and the Argonauts (side 4 at least - there's a clip of this in ld-decode-testdata)
- Red Hot and Blue
- Tubular Bells 2 (only during the programme, not the intro)
- Twister CAV

atsampson, 2022-05-22:
@harrypm observed that VITC is a lot more common [on videotape](https://github.com/oyvindln/vhs-decode/wiki/VITC-&-Subtitles), so this has more applications now in conjunction with vhs-decode.

harrypm, 2022-05-23:
@atsampson It's the primary timecode for SVHS Pro/Prosumer '90s camcorders/recorders, but recently I found quite a few LTC/VITC vice-versa injectors that could embed into any composite feed exist so technically almost anything can have VITC.

ifb, 2022-05-23:
It's just another thing that can go in the VBI and all serious tape formats that preserve at least some of the VBI can support it (2" quad is before my time and might not have, but 1" certainly does).  Making frame-accurate edits using LTC isn't nearly as reliable.

From SMPTE 12M-1-2008:

**10.6 Location of the VITC Codeword in the Vertical Interval**
The VITC codeword shall be inserted on the same line (or lines) in each field for a given analog SDTV signal.
Line numbers shown in parentheses correspond to the equivalent line in field two.

**10.6.1 525/59.94 television system**
Insertion of the address code shall not be earlier than line 10(273) or later than line 20(283). The preferred
placement of the VITC codeword is on line 14(277) and optionally on line 16(279).
If it is necessary to preserve compatibility with older equipment, VITC should appear on two nonconsecutive
lines in each field.
In such case the preferred lines are 14(277) and 16(279) except:

- lines 12(275) and 14(277) in type C recorders with sync heads;
- lines 16(279) and 18(281) in type C recorders without sync heads.

**10.6.2 625/50 television system**
Insertion of the address code shall not be earlier than line 6(319) or later than line 22(335). The preferred
placement of the VITC codeword is on television lines 19(332) and 21(334).

PR #824 (2023-01-02): Add VITC support to ld-process-vbi
The main thing here is support for extracting VITC in ld-process-vbi. I've tested this with several PAL discs and it seems to works well, but I don't have any NTSC or videotape samples - so it might need some tweaking after further testing.

In addition, I've refactored ld-process-vbi to make it easier to add new formats, and cleaned up several other warnings and oddities spotted along the way.

Fixes #474.
closed 2023-01-02 by simoninns