happycube/ld-decode #478: Video LPF vs. a real player

atsampson, 2020-04-05:
I've been experimenting with decoding the PAL Lasers disc this weekend, and found it looks better if you apply some equalisation to the TBC in order to make the VITS signals look correct -- it needed both some HF gain and some phase correction. (In addition to luma sharpness, Transform PAL will work better the flatter the frequency response is.)

This made me wonder how the frequency response in the TBC compares to the composite output of a real player. So I've used the DDD to capture both RF and composite output from my calibrated LD-V4300D for a few clips - because it's not DC coupled the composite output is a mess, but you can see the important parts of the VITS.

Science Museum Lasers (PAL, CAV, start of the disc) - all the bursts should have the same amplitude:

[image]

Jason and the Argonauts (PAL, CAV, end of the disc):

[image]

Video Essentials (NTSC, CAV, middle of the disc):

[image]

NTSC looks pretty good, but for PAL the 4.8 MHz burst is noticably attenuated in ld-decode's output relative to what the player produces...

The video LPF is an order 7 Butterworth filter with a cutoff of 4.8 MHz by default (actually applied in FFT form as usual). As an experiment, making it order 25 (!) at 5.2 MHz produces something pretty close to the real player:

[image]

But the high order means it's very non-phase-linear, which shows up in the VITS phase test - the two-frequency pulse in the middle should be symmetrical, with the bottom peaks flat at 0V:

[image]

Any ideas for a better filter profile, @happycube? (I wonder if directly generating the FFT form would work better...)

happycube, 2020-04-05:
Direct FFT filter generation would be interesting indeed.

Sounds like a higher LPF frequency is needed on PAL, possibly with a lower filter order.

atsampson, 2020-04-06:
Some non-PDO discs for comparison...

Leon (PAL, CLV, Pioneer Japan, middle of the disc):

[image]
[image]

Louvre 1 (PAL, CAV, Sonopress, middle of the disc):

[image]
[image]

The bursts are 0.5 MHz, 1.3 MHz, 2.3 MHz, 4.2 MHz, 4.8 MHz and 5.8 MHz, so it's not surprising that the last burst is mostly missing.

atsampson, 2020-04-09:
Looking at the IEC Laservision specs, an elliptic filter seems to match the specified group delay curve much better than the Butterworth does. After some searching with [this script](https://github.com/atsampson/ld-decode-testsuite/blob/master/filters/compute-video-lpf)...

[image]

PAL: `sps.ellip(N=7, rp=0.04, rs=53.0, Wn=5.3e6, fs=fs)`
NTSC: `sps.ellip(N=5, rp=0.09, rs=40.0, Wn=4.35e6, fs=fs)`

The group delay is pretty much spot on for both of those. The NTSC stopband rejection isn't very good, but it could be followed by a phase-linear LPF (a la `--WibbleRemover`). Would there be any advantage in trying to tweak those filters so one of the notches in the stopband frequency response sits on top of something (e.g. analogue audio)?

Having tried out those filters quickly in ld-decode, VITS now looks better for both PAL and NTSC, and the FFT visualisation in PAL transform2d seems to show less noise being recognised as chroma. The downside is that there's a bit more ringing on the edges of colourbars, but I'm not sure this matters much for real pictures...

Is anybody still using the "lowband" filter settings in ld-decode? It might be better to handle that with a second filter to keep the correct group delay...

atsampson, 2020-04-10:
> it could be followed by a phase-linear LPF

Or by adjusting the filter used in `scale` when converting to the output sample rate...

happycube, 2020-04-11:
I'll look at the filters tonight/this weekend.  Leaning towards pulling the PAL one in first, and then studying the NTSC one.

For group delay it would be good to factor in the entire signal processing chain - I don't *quite* know how to do that yet since it changes rather radically.

happycube, 2020-04-11:
On PAL it looks sharper, but there's a bit of ringing at the end of the high frequency test:

[image]

happycube, 2020-04-11:
This seems to do better on jasonbars for me:  

video_lpf = sps.ellip(N=7, rp=0.02, rs=100.0, Wn=5.3e6, fs=self.freq_hz) 

[image]

happycube, 2020-04-12:
Checked in some more experimental tweaks in the issue 478 branch.  Need to try it with other disks though, and it still has more ringing than I'd like.

atsampson, 2020-04-12:
The version on your branch definitely looks much better than master on Jason, Leon, Lasers and Louvre above. I also tried it on good and rotten copies of Monty Python S2, the Kagemusha leadout colourbars, and the Louvre leadin colourbars. On **all** of those the VITS multiburst and pulse look correct, and the colourbars from Jason, Lasers, Louvre and Kagemusha have about the right saturation (using the #481 branch with no compensation). Chroma/luma alignment on the Jason testcard also looks fine so the phase can't be too far off. :-)

atsampson, 2020-04-19:
I've tried the issue478 branch out on some more discs this weekend, and have found some examples that show the 5.2MHz-ish ringing in real video - I've put clips of several of these on the SFTP under adamsampson/Faults/issue478.

ahdn-vci - From the start of [A Hard Day's Night](https://www.lddb.com/laserdisc/29217/PLFEB-34311/Hard-Day's-Night-A:-The-Beatles):
[image]

smnp-kwh - From the Science Museum Nuclear Physics disc, which has an interesting mix of everything from good-quality film...
[image]

smnp-oilengine - ... to what appears to be VHS. I don't think there's much we can do about this one!
[image]

The smnp-start1/2 clips include a frame of 100% colourbars, which now (correctly) go out of range in the TBC file.

Looking at the line scope, there does appear to be quite a bit more noise with the branch - I wonder if we might do better to keep it flat up to 5 MHz and then roll off after that rather than trying to go any higher?

atsampson, 2020-04-25:
With the BPF settings from your branch, `sps.ellip(N=5, rp=0.01, rs=120.0, Wn=4.8e6, fs=self.freq_hz)` looks a bit better, and `rs=80` is nearly the same - it still rings near the cutoff but less so than the current one. Phase response (going by the tests on TCF/TCG and colourbar transitions) looks a little better as well.

I had a look to see if there were any BBC R&D reports on video LPF design... there are several! [1982-19](http://downloads.bbc.co.uk/rd/pubs/reports/1982-19.pdf) is probably the most interesting (and closest to what we're doing in a sense - 5 MHz LPF, in a system with 44 dB SNR), but [1963-35](http://downloads.bbc.co.uk/rd/pubs/reports/1963-35.pdf) definitely has the neatest pictures.

happycube, 2020-04-26:
I went back to the conventional settings and pushed it up to 5.2.  Still more rolloff in the high frequency bit, but I think it's better balanced than what's in master and could probably be merged in.

atsampson, 2022-08-20:
I think this one can be closed now - after the tweak up to 5.2MHz in 2020 it seems to be working well!

closed 2022-08-20 by atsampson