happycube/ld-decode #483: Warning from palencoder tool

simoninns, 2020-04-18:
make -j8 all shows the following warning:

palencoder.cpp: In function ‘double syncPulseGate(double, double, SyncPulseType)’:
palencoder.cpp:288:28: warning: ‘length’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     return raisedCosineGate(t, startTime, startTime + length, 200.0e-9 / 2.0);
            ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

atsampson, 2020-04-18:
No warning with current GCC or Clang, but that looks spurious anyway - the code immediately above either sets `length` or returns. For your compiler, does turning the "case NONE:" above into a "default:" make the warning go away?

simoninns, 2020-04-24:
This was my environment it seems, sorry about that... a make distclean, qmake, make all - cleared the warning.

closed 2020-04-24 by simoninns