happycube/ld-decode #489: EFM audio decoding failing for certain NTSC title(s)

putnam, 2020-05-16:
I have captured several different discs and haven't had any issues decoding EFM to PCM until I got to a specific title, [The Big Country](https://www.lddb.com/laserdisc/06521/ID7352MG/Big-Country-The:-Special-Edition).

It seems as though all three CLV sides of this title are unable to be decoded properly by ld-process-efm. While most of the audio is present, there are random missing samples spread throughout the track, creating little short gaps of silence.

Here is a Dropbox folder with a short sample. I have included the full debug output of ld-process-efm as well as everything else I could think of, minus the .lds itself. The ld-decode command I used was: ```ld-decode -n sidea.lds sidea```.

https://www.dropbox.com/sh/sbj7j3funwtlifp/AAAhWexhMhUBfEFRkWUc5qdXa?dl=0

simoninns, 2020-07-30:
Looking at the log file it seems like 9 seconds was decoded and some of the EFM was unreadable.  What is the original disc condition?  It's perfectly possible that the disc is degraded and the EFM is simply unreadable.  It might be useful to compare the SNR of the CLV disc to other CLV discs you've captured (in ld-analyse) to see if there is a significant difference in condition.

putnam, 2020-09-06:
On the exact same player, the digital audio plays back fine through the
built-in audio outputs. As far as I can tell, the discs are in pristine
condition.

(These discs have the normal audio on the digital tracks, then some extra
features on the two analog tracks)

putnam, 2020-09-06:
If it's useful, here is the SNR graph for a new capture of the same area of the disc (the first 10-12 seconds or so).

[image]

putnam, 2020-09-07:
Here is the same SNR graph for a separate CLV disc. On this capture, there were zero missing samples when I ran ld-process-efm. In both cases the sample was taken from near the beginning of the disc.

[image]

simoninns, 2020-09-16:
Related to #528

putnam, 2020-11-02:
As an update, I got a second physical copy of the same movie and it exhibits the same behavior.

atsampson, 2022-12-24:
@putnam, are you able to give this another try with current ld-decode? happycube has improved the EFM filter since this was filed, so it may work better now.

putnam, 2023-01-30:
Just re-ran ld-decode with latest trunk, and unfortunately I have the same issue.

Maybe I can provide samples so others can reproduce and confirm what I am seeing/hearing. Again as has been said this presents the same on two different physical copies and both copies look fine, and they also play back audio just fine on the player through the digital output.

Here are the first 5,000 frames as an LDF (3.8GB):
https://juarez.mantup.com/pub/bigcountry-sidea-sample.ldf

Here is the EFM I get decoding with trunk today (ld-decode -n -t16):
https://juarez.mantup.com/pub/bigcountry-sidea-efm.zip

Here is the audio I get from ld-process-efm today from the above:
https://juarez.mantup.com/pub/bigcountry-sidea-digitalaudio.flac

happycube, 2023-03-26:
Thanks, I'm looking at this one.  I don't have any specific ideas to fix it yet, but it's definitely failing in the earlier stages, probably due to filtering equialization issues and recovery tolerances.

simoninns, 2025-12-05:
Closing as this is old/stale - please retest with 2025 code and open a new issue if required