happycube/ld-decode #490: Catalogue different types of VITS signals for auto-tuning

happycube, 2020-05-24:
A long-deferred (and possibly longer-deferred) project of mine is to auto-tune not just black/white levels, but color levels and phase.  I'd like to build a set of samples of each common type of VITS for both PAL and NTSC over time.

(After #481 NTSC auto chroma gain was removed, but I'd like to reinstate - and improve - it... the burst level even with auto-MTF changes especially later in CAV disks.  I'll hopefully have examples with 12" GGV soon.)

toynbeeidea, 2020-05-27:
I don't know how/where to obtain VITS signals, but I have a lot of NTSC discs and am willing to provide... something?  I guess I need some instructions.

simoninns, 2025-12-09:
no help in 5 years - close wontfix