happycube/ld-decode #492: --WibbleRemover causes hedgehogging

Gamnn, 2020-05-25:
This clip decodes fine normally, but breaks when using --WibbleRemover:

```
ld-decode -nj issue492.ldf clip
file frame 1 CAV frame 322
file frame 2 CAV frame 323
file frame 3 CAV frame 324
file frame 4 CAV frame 325
file frame 5 CAV frame 326
file frame 6 CAV frame 327
file frame 7 CAV frame 328
file frame 8 CAV frame 329
file frame 9 CAV frame 330
file frame 10 CAV frame 331
file frame 11 CAV frame 332
file frame 12 CAV frame 333
file frame 13 CAV frame 334
file frame 14 CAV frame 335
saving JSON and exiting

ld-decode -nj issue492.ldf clip --WibbleRemover
file frame 1 CAV frame 322
file frame 2 CAV frame 323
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
file frame 3 CAV frame 324
file frame 4 CAV frame 325
file frame 5 CAV frame 326
file frame 6 CAV frame 327
file frame 7 CAV frame 328
file frame 8 CAV frame 329
file frame 9 CAV frame 330
file frame 10 CAV frame 331
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
file frame 11 CAV frame 332
file frame 12 CAV frame 333
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
file frame 13 CAV frame 334
file frame 14 CAV frame 335
saving JSON and exiting
```

[issue492.zip](https://github.com/happycube/ld-decode/files/4679189/issue492.zip)

happycube, 2020-05-26:
Looking at the line scope (line 39 is good for this) the disk already has a rather soft signal and does not need WibbleRemover.  On NTSC there are only some disks with high noise that need it.

Gamnn, 2020-05-28:
Seems you're right. A different capture of a matching disc on an HLD-X9 required --WibbleRemover and it functioned without problem, but this capture from a CLD-5000(?) didn't need it. Closing.

happycube, 2020-05-28:
That's interesting - can I see .ldf's from the same frames on both?  (Not
sure when I'll get a round tuit, but it sounds interesting!)

Gamnn, 2020-05-28:
Sure:

[5000-nowibble.ldf.zip](https://github.com/happycube/ld-decode/files/4693815/5000-nowibble.ldf.zip)
[x9-wibble.ldf.zip](https://github.com/happycube/ld-decode/files/4693817/x9-wibble.ldf.zip)

Gamnn, 2020-06-10:
Re-opening Another disc does need --WibbleRemover regardless of player, and likewise causes the hedgehogging at the start of the disc (Mostly around frame 300) for most of the captures, other than the x9's. Will PM with captures of the affected area and a wibbly area.

happycube, 2020-06-14:
Partially fixed by latest commit - the disk in question needs deemphasis reduced to .65x normal (-d .65)

Gamnn, 2020-12-06:
An update on this, rezo found the discs in this series now turn green since commit #7c6d913, unless -d .65 is specified. (even without using --NTSC_color_notch_filter/formerly --Wibble-remover)

I haven't given it a full disc decode to see if -d .65 works across the whole disc yet.

simoninns, 2025-12-09:
No progress in 5 years - close wontfix

closed 2025-12-09 by simoninns