happycube/ld-decode #493: AGC code is affected by wow/flutter

happycube, 2020-05-26:
@Gamnn found an issue where the two fields luma didn't match up after the AGC change.  Oops.

toynbeeidea, 2020-05-27:
Yeah, I think I got one too, if you need a further example.

happycube, 2020-05-27:
If it's not fixed, that'd be nice, thanks!

toynbeeidea, 2020-05-27:
Ah, yeah.  I'm afraid I'm still having an issue:  Will put a clip into this comment in the next few hours.

Gamnn, 2020-05-28:
frame 393, timecode 00:47:48.01 does appear affected.

be sure the trailing _ is in the URL if you have trouble downloading toynbeeidea's sample.

happycube, 2020-05-31:
Thanks for the tip about the _, that got me just now...

Going to look at two changes:  Changing the AGC on first fields only, and keeping a running average (which should cover wow/flutter on CLV disks as well)

toynbeeidea, 2020-06-04:
I ran the entire video through last night.  Seems better now as of 9ed93a0.  There's a bit of a luma spike or something during the cloud sequence, but I'm inclined to write it off as a flaw in the source.  Going to leave this open for the moment, as I gather that this is a partial fix?

Gamnn, 2020-06-04:
Sounds like it's due to the wow/flutter affecting the first field on one frame, causing the AGC to adjust both fields just for that frame.

In cases where there's a lot of frames like that, might be best to use the --noAGC option, until AGC is a little smarter. (maybe a rolling average, ramping up if it changes rather than spiking on single errant frames)

This is also something that might be fixed with a TBC-based AGC in once VITS is better worked out. #326 #490

toynbeeidea, 2020-06-06:
Ran it through with --noAGC, and yeah, no more luma spike - thanks!

Gamnn, 2020-06-15:
Frame 393 seems to be due to a dropout in the hsync, more examples on issue 503.

Gamnn, 2020-06-15:
Here's another interesting case. A frame goes bright with no obvious dropouts. A decode in April had an audible pop here, but no problem in the audio in the current commit. Wow/flutter related I'm guessing?

[pop.ldf.zip](https://github.com/happycube/ld-decode/files/4783441/pop.ldf.zip)

happycube, 2020-07-07:
Can you recheck this?  It should be better now.

happycube, 2020-09-27:
Checked pop.ldf.zip - doesn't sound great, but no pops, so closing

closed 2020-09-27 by happycube