happycube/ld-decode #495: ld-cut issues

toynbeeidea, 2020-05-27:
While trying to splice out a clip for #493, I used the following command:

> ./ld-cut -S 85798 -l 500  -n /run/media/erik/NAS4FreeSync/RTW.ldf /run/media/erik/NAS4FreeSync/rtw-clip.lds 

Which gave a message, "ERROR: Seeking failed," and froze until I ctrl-c'd out of it.

Backing up to 85650 gave:

> Traceback (most recent call last):
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 2697, in decodefield
>     f.process()
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 2365, in process
>     super(FieldNTSC, self).process()
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 1226, in process
>     self.linelocs2 = self.refine_linelocs_hsync()
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 1746, in refine_linelocs_hsync
>     if nb_min(hsync_area) < self.rf.iretohz(-55) or nb_max(hsync_area) > self.rf.iretohz(30):
>   File "/usr/lib/python3.8/site-packages/numba/np/arraymath.py", line 502, in array_min_impl
>     raise ValueError(MSG)
> ValueError: zero-size array to reduction operation minimum which has no identity
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "./ld-cut", line 63, in <module>
>     startloc = ldd.seek(args.start, args.seek) 
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 3126, in seek
>     fnr = self.seek_getframenr(cur)
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 3097, in seek_getframenr
>     f, offset = self.decodefield(initphase = True)
>   File "/home/erik/github/ld-decode/lddecode/core.py", line 2701, in decodefield
>     self.internalerrors.append(e)
> AttributeError: 'int' object has no attribute 'append'
> 

Backing up to 85649 worked exactly as expected.

happycube, 2020-05-28:
Try using -s to move the start point...?

happycube, 2020-05-31:
What I'll look at in the morning:  If -s is not specified (in ld-decode or -cut), try jumping the same distance.  If that fails (typically because it's a clip), try again with -s 0.

toynbeeidea, 2020-06-04:
Sorry, haven't had a chance to mess with the fix yet... will check it out shortly and report back.

toynbeeidea, 2020-07-08:
Seems to work as expected.  Nice.

closed 2020-07-08 by toynbeeidea