happycube/ld-decode #503: hsync dropout causes AGC blip

Gamnn, 2020-06-15:
Similar to #493, if there's an issue in the wrong spot, the AGC can get the brightness wrong in a frame. I am thinking AGC should be more temporally aware by default.

[image]

[agc_blip.ldf.zip](https://github.com/happycube/ld-decode/files/4781393/agc_blip.ldf.zip)

Gamnn, 2020-06-15:
Another example.

[agc_blip2.ldf.zip](https://github.com/happycube/ld-decode/files/4782184/agc_blip2.ldf.zip)

happycube, 2020-06-21:
I put in a very simple fix - using medians instead of means for each line, which should help.  I could do a bit more on this if needed though, so please test.

happycube, 2020-07-07:
Now uses regular HSYNCs to compute sync/IRE0 AGC, so this should be much more robust.

closed 2020-07-07 by happycube