happycube/ld-decode #505: hsync vroop

Gamnn, 2020-06-21:
Getting vroop in the hsync area on a couple of frames, perhaps triggered by a small dropout.
This causes the AGC to darken the frames in addition.

[image]
[image]
[image]

[vroopdark.zip](https://github.com/happycube/ld-decode/files/4810432/vroopdark.zip)

Gamnn, 2020-06-22:
This might be related. There's a long dropout at the bottom of the previous frame like above. AGC brightens the frame but no vroop.

[image]
[image]

[glitchbright.zip](https://github.com/happycube/ld-decode/files/4811057/glitchbright.zip)

happycube, 2020-07-07:
Should be fixed now...

closed 2020-07-07 by happycube