happycube/ld-decode #509: VBI/VITS showing in decoded frame

toynbeeidea, 2020-07-08:
> /home/erik/github/ld-decode/ld-decode AmonSaga-side2.ldf AmonSaga-side2 --noAGC -j -s 74

[image]

I ran this segment through my old firewire capture setup:

[image]

So it seems like there's something here that plays normally, but ld-decode doesn't like.  Just for fun I tried running ld-discmap on it, but all that did was insert a blank frame after the glitch, making it worse.

[https://drive.google.com/file/d/16um918jAYT1TG0WHafhM8mtjtcC9HjSR/view?usp=sharing](url)

Gamnn, 2020-07-10:
To be sure, have you tried re-capturing the RF?

happycube, 2020-07-12:
I've looked at this just enough to know that my latest changes actually make this issue *worse* - it never syncs up again after this frame.  

I suspect the player might've lost tracking for a moment, but I haven't gotten far enough to confirm that either way.

happycube, 2020-07-12:
OK I don't see any skips in the raw demodulation, so there's definitely a bug.  oops.

toynbeeidea, 2020-07-13:
Oh dear - so my player goofed, then?  Huh.  Okay, I guess I'll go back and try a recapture, thanks for looking!

happycube, 2020-07-14:
Reopening - the prob is current versions don't recover from this the way they ought to...

atsampson, 2020-07-14:
I think I've just found another example of this - see `adamsampson/Faults/lain9-skip-fail.ldf` on the SFTP (from PILA-1500). There's a player skip at 00:24:10.23, 5 frames in, and after that the TBC locks to the wrong vertical position for several minutes:

[image]

(I've also put `andnow-skip-fail.ldf` in there, which is another player skip that current master handles fine but was causing problems back in May - I figure this might be useful if you're testing skip recovery...)

happycube, 2020-07-15:
Committed a possible fix to branch "thereisnotry" (it was originally a cleanup branch :) ) - the failed frames in question are still vroopy, and before merging I'll add a small bit of detection/warning code.

happycube, 2020-09-27:
Should be fixed now

closed 2020-09-27 by happycube