happycube/ld-decode #510: help wanted:  good examples of poorly-decoding red pictures...

happycube, 2020-07-11:
A common LD problem that isn't solved properly in ld-decode yet is that reds on LD are very hard to get right.  So if you have any samples, I'd like to collect them to adjust filtering better someday ;)

Gamnn, 2020-07-11:
[JSLA022292-wibble.ldf.zip](https://github.com/happycube/ld-decode/files/4907113/JSLA022292-wibble.ldf.zip)

IanSmallshire, 2020-09-05:
The Image entertainment Catalog LDS file on the server (Banana1/Misc/) has a large sections of frames with varous background colours, usually a minimum of 100+ sequential frames per colour.

ld-decode <file> -s 340 for the lead-in frame
The entire disc only takes 30-40 mins to decode.

Frame background colours between: 
991- 1508 - Red
1510- 2033 - Green
2035 - 2638 - Black
2640 - 3147 - Magenta 
3149 - 3414 - Dark Yellow
3416 - 3689 - Light grey
3691 - 3869 - Cyan/Blue
3870 - 3935 - Almost white
3937 - 4135 - Orange ish...
4136 - 4237 - Dark Red
4239 - 4864 - Light Blue
4866 - 4915 - Light Green
4917 - 5000 - Pink
5002 - 5175 - Dark Grey
5177 - 5550 - Yellow ish
5552 - 5609 - Pink
5611 - 5812 - Blue

AkBKukU, 2023-11-09:
I have an RLV disc with SMPTE color bars in the lead in that may be helpful for this. I could not `ld-cut` the file down small enough to fit within github 25MB limit without `ld-decode` not being able to find frames. It's kinda janky, the VBI data is static during this. I've uploaded it to my own server to share here: https://nc.akbkuku.com/s/demTaARK6HXP5Lz and I'll keep it available for some time.
[image]

simoninns, 2025-12-09:
Closed due to inactivity

closed 2025-12-09 by simoninns