happycube/ld-decode #513: Remove or fix Comb::filterIQ?

atsampson, 2020-07-14:
`Comb::filterIQ` is the IQ post-filter in the NTSC decoder - in a conventional decoder, it would filter off aliases in the I/Q signals after demodulation.

At present, it doesn't actually do anything because of a bug in `Comb::decodeFrame`: in both places where it's called, it's applied to `currentFrameBuffer.yiqBuffer` rather than `tempYiqBuffer`. If you fix that, it doesn't work very well (the filter delay compensation needs adjusting, and its filters aren't linear-time anyway).

The NTSC decoder seems to work perfectly well without the postfilter - I guess the demodulator effectively works like a [Tayloe detector](http://www.norcalqrp.org/files/Tayloe_mixer_x3a.pdf) rather than a conventional mixer. So should I fix the bug above but have it turned off by default, or should I just remove `filterIQ` entirely?

atsampson, 2020-07-15:
Looking at the old motion detection code, that had the same problem:

```
        splitIQ(&currentFrameBuffer);

        tempYiqBuffer = currentFrameBuffer.yiqBuffer;

        adjustY(&currentFrameBuffer, tempYiqBuffer);
        if (configuration.colorlpf) filterIQ(currentFrameBuffer.yiqBuffer); // <-- here
        doYNR(tempYiqBuffer);
        doCNR(tempYiqBuffer);

        opticalFlow.denseOpticalFlow(currentFrameBuffer.yiqBuffer, currentFrameBuffer.kValues); // <-- and here
```

`splitIQ` leaves the baseband signal in `y`, so `denseOpticalFlow` was seeing the baseband signal rather than Y. That may be another reason why it never worked very well...

simoninns, 2020-08-08:
I think this is a question for @happycube  - but my vote would be 'remove'

Gamnn, 2020-08-08:
It's functional in master now.

Off:
[image]

On:
[image]

simoninns, 2020-08-08:
So, what are we actually saying here... the issue is solved and should be closed?

Gamnn, 2020-08-08:
Perhaps not entirely. It works in ld-analyse, but I don't see an option for it in ld-chroma-decoder.

atsampson, 2020-08-08:
I've made it "work" in that you can enable it now, but the filter isn't useful at the moment - the delay compensation isn't correct, and it has different delays at different frequencies (if you try it on a colour image you'll see it shifts different colours different amounts to the right). We could replace it with a linear-time FIR filter, or we could do the "filtfilt" trick (design an IIR filter with half the required amplitude, then run the data through it twice, forwards then backwards, so the delay cancels out).

simoninns, 2020-08-08:
I was trying to work out if the issue should be closed or left open (but with some indication of why) :)  You guys are not making this any easier!  Is a) the issue closed or b) not closed (please add what needs to be done) ;)

atsampson, 2020-08-08:
I think I've more or less convinced myself that fixing it (by generating a new filter) is worthwhile. I'll retitle the issue...

PR #774 (2022-08-20): Improve, and always enable, NTSC IQ filters
This is the result of some experimentation following the discussion in #513...

Comb::filterIQ now uses FIR filters generated by sps.remez, based on the ones the PAL decoder uses in 1D mode. They're a bit more expensive to compute but they're phase-linear and have less ripple and a sharper cutoff. This looks much better, to the point where I think enabling the filter now slightly improves quality in non-phase-compensating mode (and it was already always on in phase-compensating mode), so we can safely remove the options to disable it.

Narrowband Q m [... cut]
closed 2022-08-20 by happycube