happycube/ld-decode #526: Add 'disc report' feature to ld-analyse

simoninns, 2020-08-08:
Right now the analysis app produces a number of graphs and other details that show good information about the disc and it's quality.  However, comparing many discs together is quite difficult (due to the time it takes to make all the graph screen shots and the moving scales of the graphs due to differing frame lengths etc).

ld-analyse should have a report output that summarises disc condition and other details in a fixed format that's easier to compare.

If anyone has any "must have" ideas for the report, please note them below for consideration.

atsampson, 2020-08-08:
I was looking at something related to this last night, since I had a large pile of captures and decodes of them, and wanted to identify the faults to see if I could splice together a good .lds image - [this program](https://github.com/atsampson/ld-decode-testsuite/blob/master/lds-patchwork) was the result. It's not very general, but some of the faults it looks for in the JSON might be useful ideas for for what you're after:

- The decode should end with the leadout
- The VBI frame numbers should be present and follow the correct sequence (allowing for pulldown and NTSC CLV oddities, and not worrying too much about the occasional bitflip)
- seqNo should follow the correct sequence
- decodeFaults should be 0
- The field length (difference between fileLocs) should be close to the median across the disc - this is a very good indication of player skips for CLV
- The NTSC/PAL phase sequence should be correct (it doesn't check this yet)

simoninns, 2020-08-08:
It sounds like there is a lot of cross over with the existing ld-discmap tool?  Perhaps there is another 'issue' in there to do with expanding the discmapper to included some of the things in your tool (especially quality analysis) - if there is cross-over, it could be a good thing?