happycube/ld-decode #527: Support DTS Decoding

DreckSoft, 2020-08-08:
I just captured a minute of a DTS laserdisc. I was thinking as this is just the PCM track filled with DTS it might work, if I turn off error concealment but no. The issue starts quite early as all F3 frames are discarded. It won't find sync0. But the fix was very easy. Change the sync0 bit to 0x812 and that's it - DTS works. Yeah, you won't have error concealment and I'm not sure if the statistics are correct but I'd say it's a start.

I changed f3frame.cpp line from 136 to
```
    if (efmValues[0] == 0x812) {
        // Sync 0
        subcodeSymbol = 0;
        isSync0 = true;
```
Actually it's just replacing 0x801 with 0x812 in 136. Of course this will break regular PCM decoding it was just a quick hack to see if it works.

I can provide a lds or tbc sample to work with but note that the capture quality on the one I just did is far from perfect. I'm still in the process of setting everything up.

simoninns, 2020-08-09:
Are there any specifications for DTS available?

DreckSoft, 2020-08-09:
Haven't found any. It just seems regular DTS in 44100 Hz - 16bit - 1411kbps

DreckSoft, 2020-08-09:
Some general information on DTS can be found here:
https://wiki.multimedia.cx/index.php/DTS
There's also a link to a spec document (not LD specific): 
http://www.etsi.org/deliver/etsi_ts/102100_102199/102114/01.03.01_60/ts_102114v010301p.pdf

And there's libdca:
https://www.videolan.org/developers/libdca.html

DreckSoft, 2020-09-14:
I just realized that Decoding is maybe the wrong wording here. DTS processing / extraction would be more like it as the goal should be a .dts file and not a decoded .wav oder .pcm.

DreckSoft, 2020-10-11:
I've attached a patch that allows creating a DTS file in ld-process-efm if you tick "Non-standard audio decode". Basically DTS is processed the same way as PCM. The resulting file seems fine but I haven't done may checks yet. 

[dts_patch.tar.gz](https://github.com/happycube/ld-decode/files/5361179/dts_patch.tar.gz)

DreckSoft, 2021-12-02:
Any plans to merge this to the main branch? As there also seems progress with AC3RF we could get full support for all available audio formats.

DreckSoft, 2021-12-10:
Unfortunately not right now. I deleted my old test files and do not have a capture-setup right now. Maybe someone else can help out. Otherwise I'll provide a sample as soon as the equipment is set up again. Could take some time though.

DreckSoft, 2022-05-18:
I now have a new capture. Still no calibrated player but applying my patch from this thread and tick "Non-standard audio decode" will generate a file which is playable by VLC (rename it to .dts). It's 10 seconds from Armageddon (just one of the initial logos)  and it's 520MB.

Unfortunately I do not have permissions to upload to ld-decode-testdata.

atsampson, 2022-05-19:
@DreckSoft Got it, thanks! I've put a copy on Simon's SFTP.

atsampson, 2022-05-19:
Hm, that's very interesting - the current master version of ld-process-efm decodes that sample quite happily, without needing the patch above, and the resulting .pcm file plays successfully as 10 seconds of DTS 6ch audio in mpv.

Was it the same disc you tried before, when you needed the patch to get it to decode? (If so, does it only use the different SYNC0 pattern at the start of the disc, or something?)

DreckSoft, 2022-05-20:
I haven't tried it without my patch as it failed last time.

It was a different disc but I don't know which one. It was also not the start of the disc but I think also the studio logo or maybe the THX trailer. But it's strange that they use different sync words for the same format.

DreckSoft, 2022-06-13:
@atsampson 
I've uploaded another sample. This time from Goldeneye. The unpatched version won't create any dts file with that one. There's still work to be done as the F2 also seems different but at least with the patch there's an output.
https://file.io/Fw2hfQPIGdU4

I have also tried Jurassic Park Lost World - This works like Armageddon.

atsampson, 2022-08-18:
@DreckSoft - [my efmdts branch](https://github.com/atsampson/ld-decode/tree/efmdts) now has a --dts option which accepts the alternative F3 sync 0 sequence, and disables error concealment.

Is there any chance you could upload the Goldeneye sample again? I downloaded it but I can't work out where I put it - and now the link's expired! (I've still got the Armageddon one.)

DreckSoft, 2022-08-18:
@atsampson sure: https://file.io/2rCh0afMm2kW

Error concealment should also be possible for DTS by using the previous or following DTS frame. However this would require a DTS parser and won't work with the current solution.

atsampson, 2022-08-18:
Got it - thanks!

I've just updated [my branch](https://github.com/atsampson/ld-decode/tree/efmdts), and it decodes both the Armageddon and Goldeneye clips OK now. With Goldeneye, the disc doesn't have EFM timestamps, so you need to specify `--time` as well as `--dts` (common with PCM audio discs too).

atsampson, 2022-08-18:
Imminenttax provided a clip of Jurassic Park DTS, which is the same as Goldeneye - needs both `--time` and `--dts`.

PR #772 (2022-08-20): Add DTS support to ld-process-efm
This is based on @DreckSoft's work in #527. It adds a `--dts` option to accept the alternative Sync 0 value that some DTS LaserDiscs use. Since audio error concealment doesn't make sense for DTS, it also changes the default error treatment when the option is used (and checks that you haven't specified multiple treatment options).

It's been tested on clips from the following [DTS discs](https://www.lddb.com/list.php?format=ld&list=dts):
- Armageddon - decodes as normal
- Goldeneye - needs `--dts --time`
- Jurassic Park - needs `--dts --time`
- Jurassic Park: T [... cut]
closed 2022-08-20 by happycube