happycube/ld-decode #532: Don't know what to call it, just doing as asked.

toynbeeidea, 2020-08-10:
/home/erik/github/ld-decode/ld-decode Amon-Side2-RECAPPED.ldf amon-side2-new -j
ERROR:root:Unable to find any sync pulses, jumping one second
ERROR - please paste the following into a bug report:
current sample: 40000000
arguments: Namespace(MTF=None, MTF_offset=None, WibbleRemover=False, daa=False, deemp_adjust=1.0, ignoreleadout=False, infile='Amon-Side2-RECAPPED.ldf', inputfreq=None, length=110000, lowband=False, noAGC=False, nodod=False, noefm=False, ntsc=False, ntscj=True, outfile='amon-side2-new', pal=False, seek=-1, start=0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None)
Exception: need at least one array to concatenate  Traceback:
  File "/home/erik/github/ld-decode/ld-decode", line 122, in <module>
    f = ldd.readfield()
  File "/home/erik/github/ld-decode/lddecode/core.py", line 2809, in readfield
    f, offset = self.decodefield(initphase = initphase)
  File "/home/erik/github/ld-decode/lddecode/core.py", line 2780, in decodefield
    raise e
  File "/home/erik/github/ld-decode/lddecode/core.py", line 2776, in decodefield
    f.process()
  File "/home/erik/github/ld-decode/lddecode/core.py", line 2573, in process
    self.linelocs3, self.burstlevel = self.refine_linelocs_burst(self.linelocs2)
  File "/home/erik/github/ld-decode/lddecode/core.py", line 2492, in refine_linelocs_burst
    zc_bursts, field14, burstlevel = self.compute_burst_offsets(linelocs_adj)
  File "/home/erik/github/ld-decode/lddecode/core.py", line 2458, in compute_burst_offsets
    bursts_arr[True] = np.concatenate(bursts['even'])
  File "<__array_function__ internals>", line 5, in concatenate

happycube, 2020-08-10:
Can you post a small sample triggering it please?  (I'll edit the ERROR line to add that soon :) )

toynbeeidea, 2020-08-16:
Sorry for the delay, its been a loooong week.

https://drive.google.com/file/d/19cEghJv9_BqSfBb3Diie0G8mhvAyZobc/view?usp=sharing

simoninns, 2020-09-08:
@happycube - was this resolved or something done with it?  Would be good if you could update the title to the actual error and change the label.

happycube, 2020-09-17:
I'm not sure what the cause is yet... I need to look at the signal more closely, but I'll classify it as a bug for now.

happycube, 2020-10-18:
Ah... the start of a disk capture is mostly garbage data from ld-decode's perspective - there are a few viable frames but the player jumps around a lot and into the lead in and it's a huge mess.

That said, the current version has new code in the area that actually crashed, so it will try and mostly fail to decode it.

Best practice is to jump ahead several hundred frames and get an idea where the first frame is, or use "-S 1 -s 500" to use the seeking function.

happycube, 2020-10-18:
There's something else odd about this disk/cap though...

file frame 60 CLV timecode 0:00.06 frame 6
WARNING:root:Field phaseID sequence mismatch (1->1) (player may be paused)
WARNING:root:Field phaseID sequence mismatch (1->1) (player may be paused)
file frame 61 CLV timecode 0:00.07 frame 7
WARNING:root:Field phaseID sequence mismatch (1->1) (player may be paused)
WARNING:root:Field phaseID sequence mismatch (1->1) (player may be paused)
file frame 62 CLV timecode 0:00.08 frame 8
WARNING:root:Field phaseID sequence mismatch (1->1) (player may be paused)
WARNING:root:Field phaseID sequence mismatch (1->1) (player may be paused)
file frame 63 CLV timecode 0:00.09 frame 9

happycube, 2020-10-18:
Turns out the disk has _extremely_ weak higher frequency signals.  I reduced the burst threshold and got it to decode, and it cleaned up reading from the beginning more, too.  Will close today once the fix is merged into main.

commit 867272a1 (2020-10-18): fix #532
closed 2020-10-18 by happycube