happycube/ld-decode #535: Encode without deinterlacing -add note abouf field order

oyvindln, 2020-08-11:
As far as I know ffmpeg assumes bottom field first by default when encoding unless it's reading something that's flagged with field order. Video from capture devices and I think ld-decode typically ends up as top field first, so you will get some weird results when deinterlacing if its encoded the wrong way. There is a [```-top``](https://ffmpeg.org/ffmpeg.html#Advanced-Video-options)` option for specifying the output field order manually, so I suggest adding this to the wiki part about encoding without deinterlacing.

@simoninns

simoninns, 2020-08-11:
Thanks @oyvindln - I'll forward this to Stephen so he can review what needs changing on the wiki page:

https://github.com/happycube/ld-decode/wiki/Creating-video-from-PAL-decodes

simoninns, 2020-08-12:
I've updated the page with the additional information.  Thanks for the help!

closed 2020-08-12 by simoninns