happycube/ld-decode #544: Replace --WibbleRemover with a more descriptive switch

simoninns, 2020-08-29:
'Wibble' isn't a term most people would understand.  Perhaps this should be --pal-band-stop or something more meaningful?

Gamnn, 2020-08-30:
Perhaps documenting 'wibble' is the correct course here. --WibbleRemover is also used on the NTSC side of things, for distortion that looks the same as the PAL type, but the cause of the distortion and how it's mitigated is different.

happycube, 2020-08-30:
I figured I'll just change the option name in ld-decode.  I'm still not sure what a professional-y name is for it though.  Neither version is NR in the traditional sense, but that's the result.

simoninns, 2020-08-30:
iirc it's technically a notch filter?  so --notch-filter perhaps?

happycube, 2020-10-18:
Fixed.  Using --V4300D_notch_filter and --NTSC_color_notch_filter

closed 2020-10-18 by happycube