happycube/ld-decode #554: ld-decode PAL phase determination inaccurate

simoninns, 2020-09-08:
When decoding "The World the UN and You_CAV_PAL_Disc1_2019-09-15_18-27-44.raw.oga" (available from the SFTP server) there are a number of reported frames where the phase (between the fields in the frame) are out of sequence in the metadata - however visual inspection does not show any issues.

Here is the output log from ld-discmap reporting the VBI (and sequential) frames including the expected phase:

Debug: VBI Frame number 5697 ( 5696 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 8294 ( 8293 ) first and second field phases are not in sequence! - 8 expected but got 4
Debug: VBI Frame number 8296 ( 8295 ) first and second field phases are not in sequence! - 4 expected but got 8
Debug: VBI Frame number 11504 ( 11503 ) first and second field phases are not in sequence! - 8 expected but got 4
Debug: VBI Frame number 11506 ( 11505 ) first and second field phases are not in sequence! - 4 expected but got 8
Debug: VBI Frame number 11508 ( 11507 ) first and second field phases are not in sequence! - 8 expected but got 4
Debug: VBI Frame number 11509 ( 11508 ) first and second field phases are not in sequence! - 4 expected but got 6
Debug: VBI Frame number 14547 ( 14546 ) first and second field phases are not in sequence! - 4 expected but got 2
Debug: VBI Frame number 14549 ( 14548 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 14551 ( 14550 ) first and second field phases are not in sequence! - 4 expected but got 2
Debug: VBI Frame number 14553 ( 14552 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 16843 ( 16842 ) first and second field phases are not in sequence! - 4 expected but got 2
Debug: VBI Frame number 16845 ( 16844 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 16847 ( 16846 ) first and second field phases are not in sequence! - 4 expected but got 2
Debug: VBI Frame number 16849 ( 16848 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 16851 ( 16850 ) first and second field phases are not in sequence! - 4 expected but got 2
Debug: VBI Frame number 17262 ( 17261 ) first and second field phases are not in sequence! - 4 expected but got 8
Debug: VBI Frame number 17263 ( 17262 ) first and second field phases are not in sequence! - 6 expected but got 2
Debug: VBI Frame number 17265 ( 17264 ) first and second field phases are not in sequence! - 2 expected but got 6
Debug: VBI Frame number 17266 ( 17265 ) first and second field phases are not in sequence! - 4 expected but got 8
Debug: VBI Frame number 17269 ( 17268 ) first and second field phases are not in sequence! - 2 expected but got 6
Debug: VBI Frame number 17270 ( 17269 ) first and second field phases are not in sequence! - 4 expected but got 8
Debug: VBI Frame number 17273 ( 17272 ) first and second field phases are not in sequence! - 2 expected but got 6
Debug: VBI Frame number 18605 ( 18604 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 21917 ( 21916 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 24090 ( 24089 ) first and second field phases are not in sequence! - 8 expected but got 2
Debug: VBI Frame number 25685 ( 25684 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 25687 ( 25686 ) first and second field phases are not in sequence! - 4 expected but got 2
Debug: VBI Frame number 26556 ( 26555 ) first and second field phases are not in sequence! - 4 expected but got 6
Debug: VBI Frame number 27290 ( 27289 ) first and second field phases are not in sequence! - 8 expected but got 4
Debug: VBI Frame number 30508 ( 30507 ) first and second field phases are not in sequence! - 4 expected but got 6
Debug: VBI Frame number 30909 ( 30908 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 37125 ( 37124 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 37129 ( 37128 ) first and second field phases are not in sequence! - 8 expected but got 6

Note: running ld-discmap with the --maponly option is a good way to test all intra-frame field phases in a TBC.

happycube, 2020-09-27:
Got it down to three at least.  This disk has a lot of corruption in the HSYNC/burst areas used to determine field #. :(

Debug: VBI Frame number 13333 ( 13332 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 14453 ( 14452 ) first and second field phases are not in sequence! - 8 expected but got 6
Debug: VBI Frame number 28754 ( 28753 ) first and second field phases are not in sequence! - 8 expected but got 2

happycube, 2020-09-27:
All three of those are because there's a very small bit of a burst-like signal in line 6 of one field that should be silent.   Going to try tightening up the burst level tolerance from .5-1.5x median to .8-1.2x median and see if that has any issues - if not I'll commit that change and close this.

happycube, 2020-09-28:
One mistake was left after that, so I tuned the burst level tolerance some more to make .2-.8x fall back to the extrapolation-from-previous-field path.  Should get the UN disk completely right now.

happycube, 2020-10-18:
Looks good with current master, closing.

Info: Processing input metadata for un.tbc
Info: Input TBC is CAV PAL
Info: Checking for lead in and out frames...
Info: Removing 1 frames marked as lead in/out
Info: Removing invalid frames by phase analysis...
Info: Removing 0 frames marked as invalid due to incorrect phase sequence
Info: Correcting frame numbers using sequence analysis...
Info: Sequence analysis corrected 8 frame numbers
Info: Searching for duplicate frames
Info: Removed 0 duplicate frames - disc map size now 42622 frames
Info: Verifying frame numbers are present for all frames in the disc map...
Info: Verification successful
Info: Sorting the disc map into numerical frame order...
Info: Sorting complete
Info: Looking for sequence gaps to pad in the disc map...
Info: No gaps found in the disc map
Info: After padding the disc map contains 42622 frames
Info: Disc mapping process completed
Info: --maponly selected.  No output file will be written.

closed 2020-10-18 by happycube