happycube/ld-decode #555: Support PALplus Decoding

DreckSoft, 2020-09-15:
Support PALplus Decoding. Likely there's little "real life" use for that but it would be interesting to see how much improvement there is between PAL and PALplus. 

An interesting paper with examples (in German):
http://docplayer.org/166379133-Entwicklung-von-programmen-zur-simulation-eines-palplus-decoders-unter-khoros.html

ETSI specifications:
https://www.yumpu.com/en/document/view/18799852/palplus

EBU information:
https://tech.ebu.ch/docs/techreview/trev_253-analogue.pdf

Zcooger, 2020-11-23:
Probably only PAL+ LaserDisc with DTS: https://www.lddb.com/laserdisc/25319/0004/Schlafes-Bruder

DreckSoft, 2021-01-09:
Could be useful:
http://www.pix.net/mirrored/discordia.org.uk/~steve/palplus.html

@Zcooger I think it's also the only PAL LaserDisc with DTS. Unfortunately it's crazy expensive :(
I do have the regular PAL+ version though. Would be interesting to see how good this type of LD can look.

IanSmallshire, 2021-01-24:
There is now a PAL+ sample on the sftp in my area /uncalibrated/Misc/trinitron.ldf

ValZapod, 2021-05-21:
> Probably only PAL+

Nope. https://www.lddb.com/list.php?format=ld&list=palplus

DreckSoft, 2022-09-19:
I just remembered that PAL+ was used to improve analog PayTV decoding years back. 

I think the decoder just used the PAL+ helper line to speed up decoding an no real PALplus decoding was done but maybe this helps:
https://github.com/denizkoekden/Multi_Dec

DreckSoft, 2023-06-23:
This could be helpful if someone has access:
https://ieeexplore.ieee.org/document/473555

And here's another documentation:
https://www.itu.int/dms_pubrec/itu-r/rec/bt/R-REC-BT.1197-1-199802-W!!MSW-E.doc

harrypm, 2023-06-23:
> This could be helpful if someone has access:
> https://ieeexplore.ieee.org/document/473555
> 
> And here's another documentation:
> https://www.itu.int/dms_pubrec/itu-r/rec/bt/R-REC-BT.1197-1-199802-W!!MSW-E.doc

Iirc we already have these on file not 100% though.

DreckSoft, 2024-06-06:
According to the first comment below this video:
https://www.youtube.com/watch?v=t-Fk440dQsA

HackTV should have PALplus support by now. Couldn't find anything at the first glance but might be worth looking into.