happycube/ld-decode #559: Stuff ntsc3d is bad at

Gamnn, 2020-09-27:
Samples for improving ntsc3d. The CLD-R7G's 3dcomb does somewhat better on these examples, but are tricky for it as well.

dotcrawl on reds
[dotcrawl.ldf.zip](https://github.com/happycube/ld-decode/files/5287831/dotcrawl.ldf.zip)
ld-decode/R7G
[image] [image]

artifacts on shifting colors
[3dartifacts.ldf](https://drive.google.com/file/d/1H7D4ZlI0PtL5bOv2pMxADEGVGZHwZEm6/view?usp=sharing)
ld-decode/R7G
[image] [image]

atsampson, 2020-09-27:
The Sony logo is the same problem as the edge of the platter in dotcrawl.ldf:

[image]

There's a single line of colour sandwiched between two grey lines, so it's ambiguous in 2D whether it's actually colour or B/W detail - the similarity heuristic needs to be smarter about this kind of case. (If you bias it to choose the other way at the moment, then you get artefacts on colour-to-detail transitions instead.)

atsampson, 2020-09-27:
Some remaining problems in Video Essentials...

Frame 29108 - ambiguous single-line colour above "TEST":

[image]

Frame 30150ish - S&W test sequence - the diagonal blocks shouldn't have any chroma:

[image]

Frame 36430 - for one frame after a scene transition, artefacts on hair that's almost the same colour as the previous frame:

[image]

Frame 42918 - artefacts on the jacket look a bit worse in 3D than in 2D because it's otherwise sharper:

[image]

Gamnn, 2021-03-11:
Here's a tricky one, but not sure if anything can be done about it. They probably should have avoided this on something going to composite video. Crosshatch patterns really give the ntsc decoders trouble:

[image]
[checker.ldf.zip](https://github.com/happycube/ld-decode/files/6123032/checker.ldf.zip)

https://imgsli.com/NDM1NDk

Gamnn, 2021-03-13:
Here's the same issue on a different disc, though not quite as pronounced.

[image]
[checker2.ldf.zip](https://github.com/happycube/ld-decode/files/6133957/checker2.ldf.zip)

Gamnn, 2021-05-01:
Here's another with dot crawl that the adaptive filter misses.

[image]
[moredotcrawl.ldf.zip](https://github.com/happycube/ld-decode/files/6409298/moredotcrawl.ldf.zip)

cjdavis1983, 2022-12-26:
Here are some samples of dotcrawl and checkerboarding from the NTSC comb filter 
[image]
[image]

pgodwin, 2025-06-30:
Another example - DS9 2 Emissary, Part 1, Frame 2143

Source:
[image]

1d:

[image]

2d:

[image]

3d - no adaptive:
[image]

3d - adaptive:

[image]

Happy to attach a snippet of the TBC if you can point me to the right tool to do so.