happycube/ld-decode #561: Framing regression on rotter

Gamnn, 2020-10-09:
on majodemo side1 rotter, at frames 173 and 759 ld-decode reports:

WARNING:root:Field phaseID sequence mismatch (2->1) (player may be paused)

The framing then shifts from that point on.

[image]
[image]

to replicate, ld-decode -nj shift.ldf -s 500 -S 1 side1
shorter clips don't seem to reproduce

Gamnn, 2020-10-09:
Bisected and narrowed it down somewhere in here. Had to skip a lot of commits that broke entirely.

08ebc6385d3a4b62dfe427175d00fcc12e0c7d7b
fe95b641bab931a9b9f5425c90f478bf5d39012b
0cb41f6755a0eb16743330a762e82e29154caa62
979911fe0aa27a385e29d24eebf9263797a7dd5b
1bc9690d3ef11d9543be3e6735ec0ab7a60b4b24
d93a241568abe24201d729c405bf60bd993a224b
6267d9efe2ebf76f4ce19242b2db9f9c3aa5afef
b07f0d4640ef06a81c0a1ff56d3d934dccdcd7e1
e737a2bb1fcb7fc49a26e1ffe33b8b9e4ea41f33
e7a8300122280d32244cb1005137633baedd6d93
e9c9358e3b9444a2215be2ce078be4191946ad9d

happycube, 2020-10-10:
What's the command to get shift.ldf?

Gamnn, 2020-10-10:
ld-cut side1.ldf -l 1000 shift.lds

commit 6102993e (2020-10-10): fix #561 by limiting transfer of phase adjustments
closed 2020-10-10 by happycube