happycube/ld-decode #567: ld-decode fails on a disc

droark, 2020-11-09:
Tried to decode a disc and got an error as follows. This is on the latest `master` branch, Ubuntu 20.04. LDF file is ~125 GB, which doesn't seem right.

```
ERROR - please paste the following into a bug report:
current sample: 120508400
arguments: Namespace(MTF=None, MTF_offset=None, NTSC_color_notch_filter=False, V4300D_notch_filter=False, daa=False, deemp_adjust=1.0, ignoreleadout=False, infile='./INSERT_TITLE_CLV_NTSC_side1_2020-10-25_17-19-27.ldf', inputfreq=None, length=110000, lowband=False, noAGC=False, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='./The_Butterfly_Murders_Side_1', pal=False, seek=-1, start=0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None)
Exception: 'FieldNTSC' object has no attribute 'isFirstField'  Traceback:
  File "/usr/local/bin/ld-decode", line 129, in <module>
    f = ldd.readfield()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 2958, in readfield
    if len(self.fieldinfo) == 0 and not f.isFirstField:
```

droark, 2020-11-09:
Closing this out. I started around frame 500 and then reverse engineering the true first frame. That worked. I suppose this is still technically a bug, albeit one with an easy workaround.

closed 2020-11-09 by droark

Gamnn, 2020-11-09:
"-s 500 -S 1" for CAV or "-s 500 -S 0" for CLV is best practice, but I think it is planned to make it smarter about the start position later.