happycube/ld-decode #569: Override for some .json values in DD-Analyse GUI

Zcooger, 2020-11-23:
This ticket is intended to suggest adding more controls for the TBC review such as IRE adjustment in waveform monitor and active video start/end range (cropping black borders).
In case of using manual adjust original values would be still kept in .json file and overridden values used instead by the decoder.

Gamnn, 2020-11-24:
A slider for white16bIre and black16bIre would be handy, and ld-chroma-decoder should also accept the overrides as options.
Another useful tool would be eyedropper-type things to set them based on selected sample points.

I've been meaning to suggest this once #305 is implemented.

simoninns, 2020-11-29:
So far ld-analyse has been designed as a 'read only' tool for analysing the output of other tools.  I'm not convinced that it has the right 'structure' for actually editing TBC and metadata files (I found this out when trying to implement the ability to edit VBI data... it's a can of worms).

Adding a couple of sliders to modify the output of the ld-analyse viewer is possible though (and more in-line with the application's intent).  This will not update the metadata.

If that's an acceptable solution, let me know and I will modify this issue accordingly.

simoninns, 2020-11-29:
Sorry wrong button :)

Zcooger, 2020-11-29:
Keeping viewer app read-only is fine if we don't want worms to spread out.
Then I'd change the approach of the issue to "Detailed value readout in ld-analyse".
User will be responsible for keeping original .json if he wants to modify disc/tape properties in external editor.
These readouts would be:
- Current .json values of white16bIre, black16bIre, burst range and active video range.
- Cursor coordinates in waveform monitor (horizontal pixel # and 16b value) in waveform monitors.
- Cursor coordinates in TBC display (horizontal pixel #, vertical pixel #, 16b value).
- Video luminance lines in waveform monitor for limited (0-255) and full (16-235) 8b range (one of them is currently used as analog black level).
I hope that's fine explanation 👍