happycube/ld-decode #570: Average/median waveform or vectoroscope display in the DD-Analyse GUI

Zcooger, 2020-11-23:
This ticket is intended to suggest adding checkbox that enables waveform or vectoroscope display averaging for a certain amount of frames from 2 to total of TBC file.
This would allow to cancel out noise and precisely set IRE or burst level .json values while looking at VITS generator lines.

simoninns, 2020-11-29:
It's hard to understand the intent here.  If I understand correctly, you are asking for the ability to show some form of graph which is an average of something over a number of frames...  Can you be a bit more specific?  What is it (over a range of 2 or more frames) that you want to see?  If it's only for a vectorscope add the details to issue #305 and indicate that this issue can be closed as duplicate.

Zcooger, 2020-11-29:
Example of such function: https://youtu.be/KWeY2yqwVA0 - 7:28

simoninns, 2020-11-29:
The clip is an FFT function...  It still doesn't clarify what you are asking for.  You want some type of graph-based output of some type of data to animate over a specified number of frames?  Or is this some type of static average over a specified number of frames of some type of data?

Zcooger, 2020-11-29:
That "average" button enables temporal noise cancellation over time. In ld-analyse it can average current line in waveform display once decoded frames are changed back and forth. Once line is switched or "average" checkbox deselected current display might be resetted and line is displayed without filtering.