happycube/ld-decode #573: ld-process-efm:  optionally use json data for time sync on non-timecode disks

happycube, 2020-11-24:
Just had that idea while talking to smally about one of his decodes that lost sync.  The json data might not be perfectly aligned, but is probably close enough to prevent noticable loss of lipsync.

simoninns, 2020-11-29:
I don't really understand what's being proposed here... What JSON data and in what manner should it be used?

happycube, 2020-11-29:
To rephrase that, since the .json file knows (or should!) how many EFM
samples are output per field, it could be used as a timestamp for all the
80's disks that don't have timecodes.  Then, if something gets dropped it
should be able to reasonably map the output using the # of input samples
read.

This would be a fallback to prevent substantial desync, since it might be
off a bit even with this.  Hopefully this won't actually be needed after
the other audio fixes are done :)

simoninns, 2022-05-21:
Just a note for future implementation:

I think this would make more sense as an enhancement to ld-disc-mapper.  Right now it does the same processing to the analogue audio file.  If you want to remap the EFM based on the video metadata - that process is pretty much the same as what the audio code is already doing and wouldn't be too hard to adapt as an option EFM process.