happycube/ld-decode #577: Misc output requests

happycube, 2020-11-28:
smally:  Add FPS counter to frame output
gamn: Add chapter #

IanSmallshire, 2020-11-28:
Possibly have a bitwise --verbose Level on the command line where options can be configured.
1 FPS Processing speed
2 Display Chapters
4 Display % of source file processed
8 etc...

Not sure of you would want to show FPS as a real speed or take into account for 25/30fps and calculate like ffmpeg with a x0.3 fps

Zcooger, 2020-11-29:
Why not both?
For non-verbose level I'd leave as it is.
--verbose switch might provide such refreshing line:
`Processed: # ### ### ### b (###.##%), Sample frame: ### ###, RF frame: ## ###, TBC frame: ## ###, Timecode: hh:mm:ss.tt/unknown, Chapter: #, FPS: 7 / x0.3, ETA: hh:mm:ss`

happycube, 2020-11-29:
BTW, for projected output length for .ldf i'll need to add a bit to ld-ldf-reader to get the file size.  Should be doable but not quite as easy as the rest.

IanSmallshire, 2020-11-29:
Is it worth having the delimiter between the section as TAB (ASCII 9) as opposed to a space for later parsing, (not sure how shell handles this)

Another possible option that I think you have in variables:
8 - black / white SNR level output
16 - Timestamp
32 - Some form of dropout counter (total length over frame?)

etc...

Just suggestions for whats easy to implement from my point of view.

happycube, 2020-12-13:
Probably not going to finish this for rev7, but I'm going to do some output cleanup for it in any case.