happycube/ld-decode #584: erase entire line on re'draw'

happycube, 2020-12-21:
simon: "Frame 2/2: File Frame 328: CAV Frame #1 ne Lead In" - what's a 'ne lead in'?  :smile:

simoninns, 2020-12-22:
Another comment on the command line output...  It would be good if the first and last frames were summarised on a separate line also - right now it's hard to see quickly if you started the decode at the correct point (when using --start).

Error messages are also now unclear; if you get an exception, there is no indication of what frame it occurred on:

Field phaseID sequence mismatch (7->2) (player may be paused)

Field phaseID sequence mismatch (2->1) (player may be paused)

Field phaseID sequence mismatch (1->6) (player may be paused)

Field phaseID sequence mismatch (6->3) (player may be paused)

Frame 36000/110000: File Frame 36344: CAV Frame #35999

happycube, 2021-01-10:
Fixed

closed 2021-01-10 by happycube