happycube/ld-decode #588: Current frame indicator for ld-analyse graphs

Gamnn, 2020-12-28:
It would make dropouts and irregularities in SNR easier to investigate if there was an indicator on the graphs of the frame you're currently looking at in the main window.

For example:

[image]

Gamnn, 2020-12-28:
It could also possibly act as a slider, to move the main window to the desired area.

simoninns, 2020-12-30:
Needs some fairly complex enhancement to the graphs as you would also need to implement a zoom function (the red arrow in the example above is pointing to 54 fields containing the average shown).

Gamnn, 2020-12-30:
Such enhancements would be great, but the indicator being more precise than the graph shouldn't matter for this. A slider/bar overlaid on the X axis going from 0 to number_of_fields should still show the area you're looking at.

simoninns, 2021-01-04:
Fixed in master

closed 2021-01-04 by simoninns