happycube/ld-decode #601: Consider running a formatter and checker on python code

oyvindln, 2021-01-08:
Right now the code has a mix of spaces and tabs and not any consistent formatting, and when I load core.py in emacs flake8 gives up with too many errors.

I've used [black](https://github.com/psf/black) for the vhs code but there are several other alternatives. A syntax checker like flake8 could also help clean up things, get rid of unused variables and maybe even uncover bugs.

Maybe also consider splitting up core.py as it's 3k lines.

happycube, 2021-01-10:
Thanks for the pointers... I've run black in the chad-2021.01.01 branch and handled a few flake8 complaints (and will probably gradually handle more)

Yeah, core.py should be split up.  As part of the audio rework I'll put that into it's own file, and probably break out the threading bits as well.  It's almost *4*k lines now though after running through black...

happycube, 2021-01-11:
Haven't found any bugs yet, but it seriously points out clunky expressions which I'll slowly deal with ;)