happycube/ld-decode #614: Short flags for notch filters

tandersn, 2021-01-19:
Just a request for some unixy style short flags for the long flags:

--NTSC_color_notch_filter  
--V4300D_notch_filter

something like:

-nf  (ntsc notch filter)
-pf  (pal notch filter)

simoninns, 2021-12-12:
The issue here is that the short flags are too numerous and too confusing (plus there is nothing un-unix-y about longer flags).

As an example "V4300D_notch_filter" is a notch filter for the V4300D playing PAL - not a PAL notch filter.

With this in mind, I don't think short-flags add anything useful. @happycube - reopen this issue if you disagree.

closed 2021-12-12 by simoninns

tandersn, 2021-12-12:
A) what i meant, is that most unix programs have "short AND long" flags for the most common options, i didn't mean that long flags are "un unixy".  

B) "-nf  and -pf" were just vague examples, not the exact flag suggestion (whatever is appropriately decided upon ).

As an example, 'flac' has a lot more options than ld-decode, with the common ones having corresponding short flag. Just do flac --help and you can see them.