happycube/ld-decode #617: Request for CVBS (Composite Video) Decoding

titan-91, 2021-02-07:
I'm working on testing ld-decode and its child project, vhs-decode. I can't contribute any code but can provide feedback and test results with different players. This would help with testing, and give a big boost to interest in the project in general, would be cvbs/composite video decoding. I understand this may be outside the scope of the project, but this would allow me and others to compare RF decoded results to hardware playback results, while still gaining the benefits of ld-decode's unmatched TBC, comb filter, and analysis tools such as ld-analyse. The existing code for processing cvbs signals is already here, and this would have endless applications.

As far as I can tell, there is no official composite video software decoder available anywhere and people have been searching for one for years. It would be amazing if this could be implemented using the existing rate flags for 40MSPS, 35.8MSPS, and 28.6MSPS for the Domessay Duplicator and CXADC driver for the Conexant capture cards.

As far as implementation, I'm guessing it would be relatively simple. Instead of feeding the TBC code with demodulated composite video from a LaserDisc capture, just feed it the direct composite video signal from the capture, while performing DC level restoration and AGC of the signal. If there is interest in adding this feature, here's a .tar.gz archive file of some test patterns I've captured.

https://forum.videohelp.com/attachments/57098-1612232475/cvbs-samples.zip

VideoMem, 2021-02-08:
That's interesting.

Maybe a separate cvbs-tbc-decode tool/script could be added to avoid adding more program switches to the main LD workflow.

titan-91, 2021-02-21:
There's been some discussion on Discord about this, but not sure if anyone is working to implement it currently. Referencing issue #478 CVBS decoding can also help visualize filter differences and make other comparisons between RF and composite outputs for various players.

skunge2000, 2021-03-14:
D2 (and the 1/2" equivalent D3) have either SDI or Parallel Rec 656-style 4fsc digital outputs available too - removing the need to capture analogue. (The SDI variant is quite rare - it's SMPTE 259 A for 14.32MHz NTSC and B for 17.72MHz PAL from memory - with the more regular 27MHz Rec 601 SMPTE 259 C?)

The PAL standard is documented here - but electrically the NTSC variant is the same : https://tech.ebu.ch/docs/tech/tech3280.pdf

Gamnn, 2021-03-15:
#242 should help with interoperability for that.

titan-91, 2021-03-27:
The vhs-decode fork has implemented composite decoding! I'm closing this request.

https://github.com/oyvindln/ld-decode

https://ready.cs.washington.edu/~tandersn/nobackup/cvbsdecoded.png

closed 2021-03-27 by titan-91