happycube/ld-decode #621: Dvision by zero in compute_line_bursts

putnam, 2021-03-05:
I was decoding what seemed to be a very good quality capture just now and came across the following crash. I added some instrumentation and it appears that the code path for an empty burstarea is hit around line 2202, maybe last thought about in this changeset by @happycube 
https://github.com/happycube/ld-decode/commit/e737a2bb1fcb7fc49a26e1ffe33b8b9e4ea41f33

This is an NTSC disc, possibly NTSC-J. I believe it was actually authored wrong though since the source is US-based content (Looney Tunes) and the black level looks wrong at IRE 0, so I was decoding it as NTSC intentionally. The crash occurs regardless of the -j switch. Happy to share samples as well if needed.

Crash:
`
null

ERROR - please paste the following into a bug report:
current sample: 13191897799.0
arguments: Namespace(infile='speedy.lds', outfile='speedy', start=0, length=110000, seek=-1, pal=False, ntsc=True, ntscj=False, MTF=None, MTF_offset=None, noAGC=False, nodod=False, noefm=False, prefm=False, daa=False, start_fileloc=13191897799.0, ignoreleadout=False, verboseVITS=False, RF_TBC=False, lowband=False, NTSC_color_notch_filter=False, V4300D_notch_filter=False, deemp_adjust=1.0, deemp_low=0, deemp_high=0, threads=1, inputfreq=None, vbpf_high=None, vlpf=None, vlpf_order=-1)
Exception: division by zero  Traceback:
  File "/usr/local/bin/ld-decode", line 356, in <module>
    f = ldd.readfield()
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3424, in readfield
    f, offset = self.decodefield(initphase=initphase)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3403, in decodefield
    raise e
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3399, in decodefield
    f.process()
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3136, in process
    self.linelocs3 = self.refine_linelocs_burst(self.linelocs2)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3053, in refine_linelocs_burst
    field14, adjs_new = self.compute_burst_offsets(linelocs_adj)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3033, in compute_burst_offsets
    rising, phase_adjust = self.compute_line_bursts(linelocs, l)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 2697, in compute_line_bursts
    burstarea = burstarea - nb_mean(burstarea)
`

putnam, 2021-03-05:
OK, it just happened again with the other side of the same disc.

```
nulle 11994/110000: File Frame 11994: CLV Timecode CLV Timecode 27:10.05 frame 48905

ERROR - please paste the following into a bug report:
current sample: 16009658681
arguments: Namespace(infile='roadrunner.lds', outfile='roadrunner', start=0, length=110000, seek=-1, pal=False, ntsc=False, ntscj=False, MTF=None, MTF_offset=None, noAGC=False, nodod=False, noefm=False, prefm=False, daa=False, start_fileloc=-1, ignoreleadout=False, verboseVITS=False, RF_TBC=False, lowband=False, NTSC_color_notch_filter=False, V4300D_notch_filter=False, deemp_adjust=1.0, deemp_low=0, deemp_high=0, threads=8, inputfreq=None, vbpf_high=None, vlpf=None, vlpf_order=-1)
Exception: division by zero  Traceback:
  File "/usr/local/bin/ld-decode", line 356, in <module>
    f = ldd.readfield()
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3424, in readfield
    f, offset = self.decodefield(initphase=initphase)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3403, in decodefield
    raise e
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3399, in decodefield
    f.process()
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3136, in process
    self.linelocs3 = self.refine_linelocs_burst(self.linelocs2)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3053, in refine_linelocs_burst
    field14, adjs_new = self.compute_burst_offsets(linelocs_adj)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 3033, in compute_burst_offsets
    rising, phase_adjust = self.compute_line_bursts(linelocs, l)
  File "/usr/local/lib/python3.9/dist-packages/lddecode/core.py", line 2697, in compute_line_bursts
    burstarea = burstarea - nb_mean(burstarea)
```

putnam, 2021-03-05:
Head-slapping update: this occurs at the end of the capture. I captured these with DD in automated mode, with a CLV timestamp range. It seems there are empty/spurious samples at the end of the file that trigger this.

acoulte93, 2021-03-10:
This issue occurred for me on a disc, close to spin up. 

```
current sample: 383490787
arguments: Namespace(MTF=None, MTF_offset=None, NTSC_color_notch_filter=False, RF_TBC=False, V4300D_notch_filter=False, daa=False, deemp_adjust=1.0, deemp_high=0, deemp_low=0, ignoreleadout=False, infile='Disc1.lds', inputfreq=None, length=110000, lowband=False, noAGC=False, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='Disc1', pal=False, prefm=False, seek=-1, start=0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None, vlpf_order=-1)
Exception: division by zero  Traceback:
  File "/usr/local/bin/ld-decode", line 356, in <module>
    f = ldd.readfield()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3424, in readfield
    f, offset = self.decodefield(initphase=initphase)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3403, in decodefield
    raise e
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3399, in decodefield
    f.process()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3136, in process
    self.linelocs3 = self.refine_linelocs_burst(self.linelocs2)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3053, in refine_linelocs_burst
    field14, adjs_new = self.compute_burst_offsets(linelocs_adj)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3033, in compute_burst_offsets
    rising, phase_adjust = self.compute_line_bursts(linelocs, l)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 2697, in compute_line_bursts
    burstarea = burstarea - nb_mean(burstarea)
```

LDF Sample [Here](https://mega.nz/file/MnJHSYzI#Ltt3aBrd2fVeyF1XEGED7mKCaKzcMymWwcKKDSxQ6IU)

happycube, 2021-03-16:
I finally got around to looking at disc1, "It worked on my machine" with current master.  What command line settings did you use?

happycube, 2021-10-17:
AFAIK fixed