happycube/ld-decode #622: Support AC3 / AC3RF

DreckSoft, 2021-03-12:
For movie preservation, AC3 support would be helpful. First step would be to extract a proper AC3RF signal from the analog channel, second step is the demodulation. It seems not much is know on how this works. 

I found a service manual with schematics for one Demodulator:
https://manuals.lddb.com/AC3RF_Demodulators/Pioneer_RFD-1-EN_Service_Manual.pdf

Additionally, this page seems to have some information:
http://www.laaudiofile.com/ac3mod.html

DreckSoft, 2021-06-29:
There are some datasheets posted here:
https://forum.lddb.com/viewtopic.php?f=24&t=7810

And a few things are linked here:
https://forum.lddb.com/viewtopic.php?f=24&t=9612

staffanu, 2021-11-29:
I spent many late night this week trying to crack this and think I've finally succeeded. I'll write up something more complete somewhere but a summary:

I read the patent and sample the AC3-RF signal from my player. Bandpass filter at 2.88 MHZ and it is obvious by inspection that they, just as the patent says, use QPSK. There are some details I didn't quite understand, but I wrote some code to decode data. Looking at it, it is obvious that there are frames of 160 symbols, or 40 bytes. The frames are numbered 0-71, and there are 37 bytes of data per frame.

I assume there would be some kind of channel coming but  I found nothing trying brute force. So, next was trying to find an interleaving pattern for RS. The standard method (I knew nothing about this but read a few papers) seems to be to try different interleavings and check if the resulting code words are have a linear dependency between them. They did if taking every 74th byte of the data.

So, 72 frames of 37 bytes are deinterleaved as 74 code words of 36 bytes 

Assuming the code is transparent, and that no decoding has to be done except error correction turned out correct. (This is the way CIRC for EFM works). I haven't tried checking the error correcting code yet, but I did another sample of the signal, at the same time that I sampled the output from my AC3 demodulator, and wrote code to demodulate the SPDIF stream. Comparing data between my deinterleaved data and the AC3 blocks in the demodulator output it was just a tedious but satisfying task to make the puzzle to understand how the blocks are constructed. 

In short, 4 blocks of 37x72 byte input generates 5 block of 1536 bytes AC3 data. I'll write the final code to make an ac3 file from the qpsk signal so I can iron out any mistakes, but I think this is it!

atsampson, 2021-11-29:
Nice work, @staffanu!

IanSmallshire, 2021-11-29:
Great piece of detective work @staffanu 

I remember from some of the old forums there used to be a tool that either added or removed some padding on Laserdisc AC3 to make it the same as the ac3 file format but cant remember the details.

staffanu, 2021-11-29:
Yes, all of the data isn't actually used; there are some zeroes between blocks, and then there is data that isn't used. The size of that corresponds to the sub code data bandwidth, but at least my demodulator seems to skip it.

I will wrap up the code and test on a few other discs. There could be flexibility in how this padding is added, and some marking of it that I missed. If so I'm sure it can be solved somehow!

DreckSoft, 2021-11-29:
Usually people captured the Laserdisc AC3 with an SPDIF soundcard getting a "DDWAV" file. That could be converted using BeSplit:
BeSplit -core( -input input.wav -output output.ac3 -type ddwav -fix )

Maybe just using the BeSplit -fix function will remove the padding. However, decoders should ignore the padding anyways so you just save some space.

IanSmallshire, 2021-11-29:
Are there any tools for checking the integrity of an AC3 stream?
May be useful during testing!

DreckSoft, 2021-11-29:
AC3Fix for example
https://www.videohelp.com/software/AC3Fix

staffanu, 2021-11-29:
There are padding in several places. I was talking about padding in the AC3RF data to make up for the fact that the 576kbps stream, after demodation and RS decoding, has more data than the 384kbps (+32k sub channel?) data rate required for ac3. This padding has to be removed. 

There is also padding in the spdif protocol, since it is substantially higher speed. With a sample rate of 48kHz it is 1.536 kbps (and then there are quite a few bits left unused also in that -- the "raw" bit rate is 64×the sampling period, but usually only 32 bits (16 per sample) are used) a lot of padding is required. I think this is the part BeSplit does? That is all described in the IEC 61937 standard so no reverse engineering required. Anyway, that is not required in order today back the AC3 stream after capturing from AC3-RF, but is if building a hardware device to forward the data to a receiver.

staffanu, 2021-12-02:
I found the Reed-Salomon parameters and published the code at bitbucket now: https://bitbucket.org/staffanulfberg/ldaudio/src/master/

happycube, 2021-12-19:
This is awesome!  Hopefully I'll get around to adding it next year (I want to get rev7 done by the end of the holidays and don't have time to add this first.)

DreckSoft, 2022-08-14:
It seems rev7 has been postponed. Could you move this to rev 7?

happycube, 2022-08-14:
Nope, rev7 is pretty much finished, just not tagged ;)

DreckSoft, 2022-08-21:
@staffanu I read something about MUSE decoding in your repository. Just audio or also video?

staffanu, 2022-08-21:
@DreckSoft I did some preliminary work on decoding MUSE. The current state is that I built an AD converter (similar to the Domesday device) and wrote some VHDL code to synchronize with the horizontal sync on an FPGA. I captured samples at 16.2 MHz and wrote some Scala code to decode it. I extracted picture frames but I don't quite understand how inter-frame interpolation works, and spent almost no time at all on color decoding, so color is very low resolution (and also not transmission gamma corrected). I decoded the control signal (that contains the movement vector and subsampling information), and think that is correct.  Notice that I did not do any demodulation of the MUSE signal itself: I'm working on the MUSE output form the player. When (if ever) this works that could be another step but I'm not fully committed to actually make a complete decoder. Again, my interest is primarily decoding in real time. The final goal could be to build a MUSE decoder with HDMI output, or failing on the video part at least making an audio decoder that would decode MUSE and AC3 in the same box so only one connection to the amplifier.

For Audio, there is another difficulty in how to transfer 4 channel audio from the decoder to an amplifier: an obvious way is to re-code the audio into AC3 and send it over SPDIF, but I haven't found any open source AC3 encoder IP. There are open source encoders in C, of course, so it could be either ported to VHDL, or run on a soft processor in the FPGA. It is also possible to transfer uncompressed multi channel audio over HDMI (but not over SPDIF). A first version would probably just output the front channels over SPDIF.

I'm currently stuck on Audio. I did the sampling rate conversion from 16.2 MHz to 12.15 MHz, and plotting pairs of symbols on an X-Y grid shows clearly that there are 8 separate clusters of sample pairs. I used the table found in some articles to convert each pair of symbols into 3 bits, and indeed there is a repeating pattern every 1350 bits.  But it is not the synchronization pattern that the articles and books say is used. I tried de-interleaving and looked for parity of each row of 82 bits, that I think should all be the same (except for when errors occurred), but it is not.  I've been stuck at this state for a month or so now but haven't worked very intensely on it: I've tried some obvious possible misunderstandings of the encoding, such as changing the order of samples for the symbols, reversing the order of the 3 bits in the table, inverting them, etc, but no success.

If someone wants to help out on this tell me and I'll make sure to commit the latest code and I can also provide some test captures. Also just reading the relevant parts of the specification and discussing them could be useful in getting ideas about what I'm doing wrong.

The best sources on MUSE encoding I found are these:

https://archive.org/details/high-definition-television-hi-vision-technology

https://www.researchgate.net/publication/3041926_An_HDTV_Broadcasting_System_Utilizing_a_Bandwidth_Compression_Technique-MUSE

https://www.itu.int/dms_pubrec/itu-r/rec/bo/R-REC-BO.786-0-199203-W!!PDF-E.pdf

Also in Japanese, and hard copy only: https://www.amazon.co.jp/MUSE―ハイビジョン伝送方式-二宮-佑一/dp/4885520940 

Staffan

DreckSoft, 2022-08-22:
@staffanu That sounds awesome. A muse decoder with digital processing and HDMI could really improve picture quality over the 90s hardware. As for the sound: I would go for DTS encoding instead of AC3. Of course multi-channel over HDMI would be the best solution. 
There's one open source encoder: https://github.com/darealshinji/dcaenc

staffanu, 2022-08-23:
@DreckSoft Multi-channel over HDMI could be easier since no encoding is necessary. I do not know much about HDMI, however.

IanSmallshire, 2022-08-24:
@staffanu my understanding is that HDMI audio is in the vertical blanking of the video and does not have a dedicated physical channel.
There are a few people on the Discord that know more about it than I do.

staffanu, 2022-09-11:
Just to say I finally solved the 2 symbols to 3 bit conversion and see the synchronization pattern in the MUSE audio data. It turns out that the table in the ITU recommendation and also in one other publication is incorrect. The table in the Japanese book is correct, however, or at least produces the frame synchronization code!

happycube, 2022-12-25:
This still needs work on some disks, but I'm going to close it anyway and open another issue later :)

closed 2022-12-25 by happycube