happycube/ld-decode #625: Error when using ld-decode

ArcadeHacker, 2021-03-28:
Ubuntu 18.04  / Python 3.6.9

$ld-decode --PAL RF-Sample_2021-03-27_23-29-05.lds outputfile

ERROR - please paste the following into a bug report:
current sample: 551576.769231
arguments: Namespace(MTF=None, MTF_offset=None, NTSC_color_notch_filter=False, RF_TBC=False, V4300D_notch_filter=False, daa=False, deemp_adjust=1.0, deemp_high=0, deemp_low=0, ignoreleadout=False, infile='RF-Sample_2021-03-27_23-29-05.lds', inputfreq=None, length=110000, lowband=False, noAGC=False, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='outputfile', pal=True, prefm=False, seek=-1, start=0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None, vlpf_order=-1)
Exception: Failed at nopython (nopython mode backend)
cannot convert native ?float64 to Python object  Traceback:
  File "/usr/local/bin/ld-decode", line 356, in <module>
    f = ldd.readfield()
  File "/usr/local/lib/python3.6/dist-packages/lddecode/core.py", line 3428, in readfield
    f, offset = self.decodefield(initphase=initphase)
  File "/usr/local/lib/python3.6/dist-packages/lddecode/core.py", line 3407, in decodefield
    raise e
  File "/usr/local/lib/python3.6/dist-packages/lddecode/core.py", line 3403, in decodefield
    f.process()
  File "/usr/local/lib/python3.6/dist-packages/lddecode/core.py", line 2910, in process
    self.fieldPhaseID = self.determine_field_number()
  File "/usr/local/lib/python3.6/dist-packages/lddecode/core.py", line 2863, in determine_field_number
    rising, phase_adjust = self.compute_line_bursts(self.linelocs, l)
  File "/usr/local/lib/python3.6/dist-packages/lddecode/core.py", line 2719, in compute_line_bursts
    burstarea, 0, len(burstarea) - 1, threshold
  File "/usr/lib/python3/dist-packages/numba/dispatcher.py", line 307, in _compile_for_args
    return self.compile(tuple(argtypes))
  File "/usr/lib/python3/dist-packages/numba/dispatcher.py", line 579, in compile
    cres = self._compiler.compile(args, return_type)
  File "/usr/lib/python3/dist-packages/numba/dispatcher.py", line 80, in compile
    flags=flags, locals=self.locals)
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 740, in compile_extra
    return pipeline.compile_extra(func)
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 360, in compile_extra
    return self._compile_bytecode()
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 699, in _compile_bytecode
    return self._compile_core()
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 686, in _compile_core
    res = pm.run(self.status)
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 246, in run
    raise patched_exception
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 238, in run
    stage()
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 621, in stage_nopython_backend
    self._backend(lowerfn, objectmode=False)
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 576, in _backend
    lowered = lowerfn()
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 563, in backend_nopython_mode
    self.flags)
  File "/usr/lib/python3/dist-packages/numba/compiler.py", line 860, in native_lowering_stage
    lower.create_cpython_wrapper(flags.release_gil)
  File "/usr/lib/python3/dist-packages/numba/lowering.py", line 229, in create_cpython_wrapper
    release_gil=release_gil)
  File "/usr/lib/python3/dist-packages/numba/targets/cpu.py", line 149, in create_cpython_wrapper
    builder.build()
  File "/usr/lib/python3/dist-packages/numba/callwrapper.py", line 122, in build
    self.build_wrapper(api, builder, closure, args, kws)
  File "/usr/lib/python3/dist-packages/numba/callwrapper.py", line 176, in build_wrapper
    obj = api.from_native_return(retty, retval, env_manager)
  File "/usr/lib/python3/dist-packages/numba/pythonapi.py", line 1325, in from_native_return
    out = self.from_native_value(typ, val, env_manager)
  File "/usr/lib/python3/dist-packages/numba/pythonapi.py", line 1339, in from_native_value
    return impl(typ, val, c)
  File "/usr/lib/python3/dist-packages/numba/targets/boxing.py", line 442, in box_tuple
    obj = c.box(dtype, item)
  File "/usr/lib/python3/dist-packages/numba/pythonapi.py", line 56, in box
    return self.pyapi.from_native_value(typ, val, self.env_manager)
  File "/usr/lib/python3/dist-packages/numba/pythonapi.py", line 1336, in from_native_value
    raise NotImplementedError("cannot convert native %s to Python object" % (typ,))

happycube, 2021-03-28:
Ubuntu 18.04 support was dropped in November, current versions have a numba optimization not supported in the version there.  (Sorry.)

closed 2021-03-28 by happycube

oyvindln, 2021-03-28:
You can disable the numba JIT completely with the NUMBA_DISABLE_JIT environment variable, e.g `NUMBA_DISABLE_JIT=1 ld-decode --PAL RF-Sample_2021-03-27_23-29-05.lds outputfile,` which may prevent the error, though that will cause ld-decode to be extremely slow.

ArcadeHacker, 2021-03-28:
Thank you guys, installing Ubuntu 20 instead.