happycube/ld-decode #632: Remove "75% white point" option?

atsampson, 2021-04-14:
ld-chroma-decoder currently has an option `-w` to use 75 IRE as the white level. It looks like this was the default at some point in the past (#113 changed it).

Are there actually discs deliberately mastered with 75 IRE white, or can this be removed now?

ifb, 2021-05-17:
I'm fine with removal.

ValZapod, 2021-05-21:
What does it mean 75% white level? Reference white is 100 IRE, superwhite is possible untill 133 IRE, 120 or 130 IRE, depending on the system in use. https://zone.ni.com/reference/en-XX/help/370524V-01/siggenhelp/video_levels/

ifb, 2021-05-21:
It means to pretend that 75% white = 235 in 8-bit YCbCr.  That's obviously not a thing, which is why it's slated for removal.  If 75% white is a useful thing, nobody has spoken up about it.

Oh, and LD is analog.  Nothing is reserved.  There's no concept of that.  You can (poorly) master the disc and make downstream devices (like an RF modulator) deal with it.

ValZapod, 2021-05-21:
> Oh, and LD is analog. Nothing is reserved.

That should be false. Blanking/pedestal, sync, close captioning in 21 line or its equivalent. And anyway you present it on modern displays, they may support supewhite if YCbCr passthrough is done (like my LG C9) but 255 and 0 are reserved in limited range formats, it may be supported in hevc and avc annex b but not in HDMI.

>pretend that 75% white = 235 in 8-bit YCbCr.

And what that will be for black level and chroma? 

> Edit after discussion on IRC: Even if there are

ifb, 2021-05-21:
This has nothing to do with anything you're talking about.  The 75% white option simply changes a scaling factor.  It's like cranking the gain/brightness control way up.

ValZapod, 2021-05-21:
> The 75% white option simply changes a scaling factor

You mean quntisation from float? You actially float conversion itself?

Are there any examples of such mastered disks?

PR #643 (2021-05-30): Remove the "75% white point" option
This appears to be left over from early development; it's not very useful for real-world decoding.

Fixes #632.
closed 2021-05-30 by atsampson