happycube/ld-decode #636: ld-decode failure

droark, 2021-04-30:
Hi. I was attempting to decode a disc and got a failure near the end. Here are the details.

```
ERROR - please paste the following into a bug report:
current sample: 136676423751.17174
arguments: Namespace(MTF=None, MTF_offset=None, NTSC_color_notch_filter=False, RF_TBC=False, V4300D_notch_filter=False, daa=False, deemp_adjust=1.0, deemp_high=0, deemp_low=0, ignoreleadout=False, infile='./In the Line of Duty_CLV_NTSC_side1_MeiAh_2021-04-29_06-22-32.ldf', inputfreq=None, length=110000, lowband=False, newaudio=False, noAGC=False, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='./D_Side1', pal=False, prefm=False, seek=-1, start=0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None, vlpf_order=-1)
Exception:   Traceback:
  File "/usr/local/bin/ld-decode", line 398, in <module>
    f = ldd.readfield()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3432, in readfield
    f, offset = self.decodefield(initphase=initphase)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3411, in decodefield
    raise e
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3407, in decodefield
    f.process()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3123, in process
    super(FieldNTSC, self).process()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 1451, in process
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 2117, in compute_linelocs
    line0loc, lastlineloc, self.isFirstField = self.getLine0(validpulses)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 1967, in getLine0
    limit = 100 if (self.prevfield is not None and self.prevfield.skip_check() >= 50) else None
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 1934, in skip_check
    line_ire = self.rf.hztoire(nb_median(self.data['video']['demod'][sl]))
  File "/usr/lib/python3/dist-packages/numba/targets/arraymath.py", line 1079, in _select_two
    assert high > low  # by construction
```

atsampson, 2021-05-24:
That's a strange failure - it's actually an internal check inside numba's implementation of numpy.median failing (which means it's either a numba bug, or a random hardware bitflip). Are you able to reproduce it with a short clip?

simoninns, 2021-12-12:
No feedback in months.  Closing.

closed 2021-12-12 by simoninns

droark, 2021-12-12:
Ack! Sorry about that. Completely forgot about this issue, to be honest. I'll see if I can remember which file it was and try again on the latest `master`. If I still have an issue, I'll try with a short clip and open a new issue if I still get a crash.