happycube/ld-decode #642: fix pickle problem on macOS

Gamnn, 2021-05-18:
ld-decode is currently broken on macOS if using python ≥ 3.8.

Due to a macOS issue, from python 3.8 the start process defaults to 'spawn', which breaks the current ld-decode multiprocessing code.
https://docs.python.org/3/library/multiprocessing.html#contexts-and-start-methods

Best solution would be to refactor to allow either fork or spawn, but it can be worked around by appending `, set_start_method` to `from multiprocessing import Process, Queue, JoinableQueue, Pipe`

and adding `set_start_method('fork')` before `self.q_in = JoinableQueue()`

droark, 2021-05-25:
I can confirm that this workaround does work for macOS. I can draw up a PR later that will be macOS-only.

droark, 2021-07-24:
https://github.com/happycube/ld-decode/pull/653

happycube, 2021-07-28:
Thanks droark - (finally) merged that - does that fix this for you Gamn?

Gamnn, 2021-07-29:
Yep, seems fine!

closed 2021-07-29 by Gamnn