happycube/ld-decode #647: ld-cut failing

IanSmallshire, 2021-07-15:
Command run below with filenames removed:

Maybe related to #645 

ian@ian-ddd230:/media/ian/NAS/Banana1/Kal$ ld-cut Infile.lds outfile -s 2400 -l 100 --pal
Traceback (most recent call last):
  File "/usr/local/bin/ld-cut", line 102, in <module>
    ldd = LDdecode(filename, None, loader, system=system, doDOD=False, _logger=logger)
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 3233, in __init__
    self.rf = RFDecode(
  File "/usr/local/lib/python3.8/dist-packages/lddecode/core.py", line 328, in __init__
    deemp_low, deemp_high = extra_options["deemp_coeff"]
KeyError: 'deemp_coeff'
ian@ian-ddd230:/media/ian/NAS/Banana1/Kal$ ld-cut --help
usage: ld-cut [-h] [-s start] [-l length] [-S seek] [-E end] [-p] [-n] [-C LDFCOMP] infile outfile

atsampson, 2021-07-15:
We should have a test for ld-cut in the testsuite to catch this kind of thing...

IanSmallshire, 2021-07-15:
Confirmation that this is fixed and I have now cut the file. 
Is this now left open for a test to be added?

happycube, 2021-07-18:
Yeah, there should be an added test for ld-cut.  This could be closed though.

closed 2021-12-12 by simoninns